MagicMirrorOrg/MagicMirror #3227: Weather module (yr) hung on 'Loading ...'

Justheretoreportanissue, 2023-10-11:
Hi!

Using default weather module with provider 'yr' on MM v2.25.00

Is it just me or is there a problem with the api for 'sunrise'?

In the MM all 3 of my weather modules are stuck on 'Loading ...' since yesterday.

In the Console I can see the following errors:
`Could not load weather data. SyntaxError: Unexpected non-whitespace character after JSON at position 4 (line 1 column 5)
    at JSON.parse (<anonymous>)
    at performWebRequest (utils.js:27:29)`
    
Digging a bit deeper into the Chrome Console I find that the request for:
`https://api.met.no/weatherapi/sunrise/2.3/sun?lat=<XX.XXXX>&lon=<YY.YYYY>&date=2023-10-11&offset=%2B02%3A00`

Gives a response of:
`404 Not Found`
`The specified version number is not supported by this product`

I get the same error when trying the above with curl in a terminal on the MM server.

Apparently the api for 'locationforecast' still works.

Running this with curl in a terminal produces nice json output:
`https://api.met.no/weatherapi/locationforecast/2.0/complete?&altitude=1&lat=<XX.XXXX>&lon=<YY.YYYY>`

*) Yes, I have exchanged my 'lat' and 'lon' with placeholders.

Regards,

   /Jonas...

Justheretoreportanissue, 2023-10-11:
Answering myself here...

Could it be that v2.3 of this api is no longer supported and that v3.0 should be used instead?

`https://api.met.no/weatherapi/sunrise/3.0/sun?lat=<XX.XXXX>&lon=<YY.YYYY>&date=2023-10-11&offset=%2B02%3A00`

Regards,

   /Jonas...

khassel, 2023-10-11:
They are fast with their api changes ...

We should introduce `apiVersionSunrise` and `apiVersionLocationforecast` so users can change this in `config.js`.

> Could it be that v2.3 of this api is no longer supported and that v3.0 should be used instead?
> https://api.met.no/weatherapi/sunrise/3.0/sun?lat=<XX.XXXX>&lon=<YY.YYYY>&date=2023-10-11&offset=%2B02%3A00

Did you test this statement? Is the output valid json?

If so you could change the version from 2.0 to 3.0 in [this line](https://github.com/MichMich/MagicMirror/blob/343e7de7bd295a69e68e4ee520552e0785a99e1c/modules/default/weather/providers/yr.js#L198) and test if the module works with 3.0

Justheretoreportanissue, 2023-10-12:
Was about to get back here and confirm it working.

Have already gone through the code and made that exact change.

Yes, confirmed working after the change. No more problem with 'Loading ...'

Regards,

/Jonas...

rejas, 2023-10-12:
Thx for the report. Do you want to create a PR for this fix too?

Justheretoreportanissue, 2023-10-12:
Sorry, not that smart with GH, just able to read and correct code.

Please feel free to make the correction and sort it out. I just do not have that skillset (yet).

/Jonas...

rejas, 2023-10-12:
Ok, will create a PR later for this.

johnhelt, 2023-10-23:
The default values for the forecastApiVersion is 3.0 by default, but it should be 2.0.

rejas, 2023-10-23:
Is 2.0 working?

johnhelt, 2023-10-23:
Yes for the forecast api.

rejas, 2023-10-23:
Thx for the info!

PR #3223 (2023-10-15): Fix API version in yr weather provider call
Fixes #3227 

and also
- removes unused code
- de-duplicates code fragments
- fixes typos
- inlines code
- adds more weather util tests

@martingron and @Justheretoreportanissue would you be so kind to check if I didnt mess anything up?
commit a0b444d6 (2023-10-15): Fix API version in yr weather provider call (#3223)

Fixes #3227 

and also
- removes unused code
- de-duplicates code fragments
- fixes typos
- inlines code
- adds more weather util tests

@martingron and @Justheretoreportanissue would you be so kind to check
if I didnt mess anything up?
PR #3248 (2023-10-23): Fix yr weather provider api version
Fixes #3227 once more
commit 04f0df26 (2023-10-23): Fix yr weather provider api version (#3248)

Fixes #3227 once more
closed 2024-01-01 by MichMich