MagicMirrorOrg/MagicMirror #3228: Documentation error...

BKeyport, 2023-10-12:
Was poking around the documentation and noticed a broken link: 

Starting at https://docs.magicmirror.builders/development/core-module-file.html#this-updatedom-speed-options

if you click on animate guide in the table, it'll go to https://docs.magicmirror.builders/modules/animate#animatein or https://docs.magicmirror.builders/modules/animate#animateout and report "No input file specified"

rejas, 2023-10-12:
Thx for the report, do you want to fix it with a PR?

BKeyport, 2023-10-12:
I don't know how the documentation system works, and would prefer that if
the page is around somewhere, it could be fixed by someone knowledgeable.
-- B

rejas, 2023-10-12:
ok, will create a PR myself then.

bugsounet, 2023-10-12:
Hi, 

Thanks for this report.

I will correct each link this evening

Note:
I think vue press is not up-to-date because in local it's works
@MichMich

bugsounet, 2023-10-12:
@rejas if you do it perfect ;)