MagicMirrorOrg/MagicMirror #3229: Strange error messages and MMM-Fuel won't sort properly

dyeatman, 2023-10-12:
Platform (Hardware/OS): Raspberry Pi 3A+
Node version: 18.18.0
MagicMirror² version: 2.25.0
Module version: 2.4.0

Everything on my MM works fine.
The only issue ( fewieden can't explain) is the MMM-Fuel sortBy option works for him but not for me. 

I the process of t/s to find the problem in MMM-Fuel I see strange error messages in my mm-error.log:

```
| [1442:1012/141354.525010:ERROR:object_proxy.cc(590)] Failed to call method: org.freedesktop.portal.Settings.Read: object_path= /org/freedesktop/portal/desktop: org.freedesktop.DBus.Error.ServiceUnknown: The name org.freedesktop.portal.Desktop was not provided by any .service files
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | ../../buildtools/third_party/libc++/trunk/include/vector:1473: assertion __n < size() failed: vector[] index out of bounds[1442:1012/141409.530401:ERROR:gpu_process_host.cc(957)] GPU process exited unexpectedly: exit_code=134
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | ../../buildtools/third_party/libc++/trunk/include/vector:1473: assertion __n < size() failed: vector[] index out of bounds[1442:1012/141419.733617:ERROR:gpu_process_host.cc(957)] GPU process exited unexpectedly: exit_code=134
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | MESA-LOADER: failed to retrieve device information
0|mm       | ../../buildtools/third_party/libc++/trunk/include/vector:1473: assertion __n < size() failed: vector[] index out of bounds[1481:1012/141428.255867:ERROR:command_buffer_proxy_impl.cc(128)] ContextResult::kTransientFailure: Failed to send GpuControl.CreateCommandBuffer.
0|mm       | [1442:1012/141428.477959:ERROR:gpu_process_host.cc(957)] GPU process exited unexpectedly: exit_code=134
0|mm       | [1481:1012/141432.847298:ERROR:command_buffer_proxy_impl.cc(128)] ContextResult::kTransientFailure: Failed to send GpuControl.CreateCommandBuffer.
```

khassel, 2023-10-12:
This repo is about mm core and the default modules, we cannot support foreign modules here.

I never saw such assertion errors, one thing you can try is do delete the `~/config/Electron` folder.

dyeatman, 2023-10-12:
Apologies if this is the wrong place….

When comment out ALL the modules (including the default ones) I still get the errors in the log…so it seemed to be a core issue.

The MMM-Fuel author says he thought it was a core MM issue…not his module since it worked fine for him. 

At his suggestion I updated Electron.

I will delete the Electron folder as suggested.

khassel, 2024-01-03:
no feedback, nothing reported from others so closing.

closed 2024-01-03 by khassel