MagicMirrorOrg/MagicMirror #3232: MagicMirror crashes since update

Rene1695, 2023-10-15:
Hi all,

from one to the other day my magic mirror stops working completely. I was not able to access the webpage anymore so I made an update to the latest version but I encountered the same problem. 

I am running magic mirror on docker on my unpaid server and it was working almost a year without any issue. Now the strange thing is that the docker is started successfully but once I got into the console, I got a "reconnected" message every 10 seconds. Also when looking into the log, something is going on and I can see the log but then after a couple of seconds the log is crashing (the log window just closes). To capture the log I made a screen video. Anybody has an idea on this? 

[image]

sdetweil, 2023-10-15:
looks like the npm install did not complete correctly, or you don't have the right version of nodejs

do
```
node -v
```
show the results

how did u upgrade
what device and os?
output of these commands 
uname -a
lsb_release -a
free -m

Rene1695, 2023-10-15:
node -v 
v18.17.1

uname-a
Linux bbdfb355be3a 5.19.17-Unraid #2 SMP PREEMPT_DYNAMIC Wed Nov 2 11:54:15 PDT 2022 x86_64 GNU/Linux

leb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 11 (bullseye)
Release:        11
Codename:       bullseye

free -m
               total        used        free      shared  buff/cache   available
Mem:           15898        2165        8894        1915        4838       11525
Swap:              0           0           0

I upgrade via Unraid, there is just a button where he pulls the newest update from GitHub. But I think the error was there before. Also while entering these commands the console was very glitchy. Several reconnects and I needed to reenter the command. There is something weird going on. But I didn't changed anything on my server.

khassel, 2023-10-15:
from where did you start the commands above, from your host or from withing the docker image? Which docker image are you using?

Rene1695, 2023-10-15:
> from where did you start the commands above, from your host or from withing the docker container? Which docker image are you using?

Within the docker console. When I tried this in my server console I got "node: bash: command not found". I am using the latest docker image (v.2.25.0)

Rene1695, 2023-10-15:
Is there a way to install a previous version? Because in docker hub I only see a tag for the latest version.

khassel, 2023-10-15:
> Which docker image are you using?

still not answered ... provide a link to docker hub

Rene1695, 2023-10-15:
Ah sorry I am stupid.

docker -v 
Docker version 20.10.21, build baeda1f

khassel, 2023-10-15:
> Because in docker hub I only see a tag for the latest version.

I asked about the image name, tag is `latest` but which image are you using?

Rene1695, 2023-10-15:
Ok than I am not sure what you mean, sorry. This is my config in unraid for the docker container. This is the link to the docker hub it uses.

[image]

https://hub.docker.com/r/ich777/magic-mirror2

khassel, 2023-10-15:
this seems to be a special setup for unraid, we don't support this here.

From the error message it looks like you are using an older version of the default calendar module with the latest mm version.

At the end of the above link is a support thread mentioned, you have to ask there.

khassel, 2023-10-15:
concerning the docs of the docker image there are params

<html><body>

FORCE_UPDATE | This will force an update of the MagicMirror² and also the modules | ""
-- | -- | --
FORCE_UPDATE_MODULES | This will force an update of the modules for MagicMirror² | ""

</body>
</html>

so may setting them will fix your issue.

As this is unrelated to mm core I will close this here.

closed 2023-10-15 by khassel