MagicMirrorOrg/MagicMirror #3233: Compliments not loading from local file.

jschneekloth, 2023-10-15:
My docker compose file has volumes setup as:
```
    volumes:
      - ~/volumes/config:/opt/magic_mirror/config
      - ~/volumes/modules:/opt/magic_mirror/modules
      - ~/volumes/css:/opt/magic_mirror/css
```

I copied `compliments.json` which is straight copy of [this compliments](https://gist.githubusercontent.com/ryck/e28a69665b8839f6e9a7acd6b4acc97d/raw/be1dee8f805a433f6ee0fa3556d1927da14e7799/compliments.json) file to the modules directory.  My config for complements is setup as follows:
```
{
      module: "compliments",
      position: "lower_third",
      config: {
        remoteFile: "/opt/magic_mirror/modules/compliments.json",
     },
}
```

When ran, the default compliments are shown and not the remoteFile that I've specified above.  Here's output of running `docker compose exec -t -i magic-mirror ls /opt/magic_mirror/modules` on the host:
```
MMM-IQAir  MMM-WiFiPassword  calendars  compliments.json  default
```

When I try two other versions of this configuration, the compliments load as expected:
```
{
      module: "compliments",
      position: "lower_third",
      config: {
        remoteFile: "[/opt/magic_mirror/modules/compliments.json](https://gist.githubusercontent.com/ryck/e28a69665b8839f6e9a7acd6b4acc97d/raw/be1dee8f805a433f6ee0fa3556d1927da14e7799/compliments.json)",
     },
}
```
^^^ Works

```
{
      module: "compliments",
      position: "lower_third",
      config: {
        compliments:  {
             // Paste of content of above gist
        }
     },
}
```
^^^ Works

Any idea on what might be going on here, why can't I reference the local file for compliments?

sdetweil, 2023-10-15:
the remotefile contains only everything after the compliments:
in aon format
```
{
...
}
```

json has all text in double quotes

there are some good online validators

jschneekloth, 2023-10-15:
I corrected one of the working examples above, bad copy/paste.  @sdetweil not sure what you're calling out specifically, sorry if that caused confusion above.  The referenced gist works if referenced via URL, but if I instead bring that file local to the container, it doesn't load as expected and instead the default compliments are displayed.  I don't think this is a json formatting issue as the same contents are present without modification in the gist as well as the json file presented from local file system.

khassel, 2023-10-15:
you can't use the whole filename from the underlying file system because the web server does load this within the `compliments.js` file.

So you have to use a relative path, in this case the `compliments.json` is 2 directories above so

```js
remoteFile: "../../compliments.json"
```

should work.

jschneekloth, 2023-10-15:
@khassel That did the trick, and makes sense, thanks!

closed 2023-10-15 by khassel