MagicMirrorOrg/MagicMirror #3234: Installation Issue with node. js

donblaylock, 2023-10-15:
I have cloned MagicMirror but when I run install using 
npm run install-mm
I get a message that nodejs needs to be >=18
The latest version I can get is 16.20.2
Also I am on npm version 8.19.4
Can anyone please help. I’ve been working on this all day with no success.

khassel, 2023-10-15:
there are enough issues about this in the forum, you can use @sdetweil scripts to handle this (you find the link in the installation section of the docs).

donblaylock, 2023-10-15:
Thanks for the info

Leto95, 2023-10-26:
Is it this:
https://github.com/sdetweil/node-ical

sdetweil, 2023-10-26:
@Leto95   no
see 
https://github.com/sdetweil/MagicMirror_scripts

Leto95, 2023-10-28:
Thank you, but node process prevents the upgrade and cannot be terminated with killall

Node should be upgraded.
A Node process is currently running. Can't upgrade.
Please quit all Node processes and restart the update.
running process(s) are
schneew+ 3873 3872 0 00:33 ? 00:00:00 node ./node_modules/.bin/electron js/electron.js

sdetweil, 2023-10-28:
you must stop the running MagicMirror..

pm2 stop all

sdetweil, 2023-10-29:
I just uploaded an enhancement that will detect pm2 and shut down mm for you

Leto95, 2023-11-13:
Thank you so much. But when i install nodejs, a message told me 20.5.1 is the newest version. I'm confused.
Do you know why I get this message?

sudo apt install nodejs
Paketlisten werden gelesen... Fertig
Abh?ngigkeitsbaum wird aufgebaut... Fertig
Statusinformationen werden eingelesen... Fertig
nodejs ist schon die neueste Version (20.5.1-deb-1nodesource1).

sdetweil, 2023-11-13:
cause its a mess

uninstall it 

sudo apt-get remove nodejs

then use my install script
see https://github.com/sdetweil/MagicMirror_scripts

sdetweil, 2023-11-25:
@Leto95   are you running ok now?

Leto95, 2023-11-26:
@sdetweil Yes, I was able to solve the problem now. It wasn't always clear which script to use.

sdetweil, 2023-11-26:
@Leto95 can u tell me more about the lack of clarity. I'd like to fix the doc to help others.

closed 2024-01-18 by khassel