MagicMirrorOrg/MagicMirror #3235: [newsfeed] Line break tags <br> should be replaced with spaces

d42ohpaz, 2023-10-16:
As an example, if you look at the RSS feed for lemmy.world, you'll notice that it uses line breaks. Because the newsfeed module seems to strip out HTML (rightfully so), it causes some of the text to be butted up against each other:

[image]

Instead of stripping out <br> tags, it should replace them with a space (or non-breaking space (&nbsp;)).

bugsounet, 2023-10-16:
Hi, I don’t know if that’s really intended but according to the code it removes the `<...>` tag and replaces it with nothing.

see original code in `~/MagicMirror/modules/default/newsfeed/newsfeedfetcher.js` [line 56](https://github.com/MichMich/MagicMirror/blob/master/modules/default/newsfeed/newsfeedfetcher.js#L56)

Try to replace this line by `description = description.toString().replace(regex, " ");`

maybe it will be better in your case

d42ohpaz, 2023-10-17:
@bugsounet Yes, that's exactly the solution needed.

MichMich, 2023-10-17:
I've got absolutely no idea anymore. 🙈

rejas, 2023-10-17:
> I've got absolutely no idea anymore. 🙈

well, 7 years is a long time... but maybe its really just trying to get rid of any html tags. which looks weird with <br> in the text. 

would just replacing those with an empty space " " look always okay I wonder?

@d42ohpaz do you want to code something (remove the regex and try to make a good looking description with some more options passed to the html-to-text library)?

bugsounet, 2023-10-17:
I will inspect  html-to-text library options because actually, without regex, It display web links. I don't think it's raeally neded to display this

rejas, 2023-11-11:
@d42ohpaz can you check if the latest develop branch fixes your problem?

d42ohpaz, 2023-11-20:
> @d42ohpaz can you check if the latest develop branch fixes your problem?

Yup, looks fixed in latest develop branch. Thank you!

closed 2024-01-01 by khassel