MagicMirrorOrg/MagicMirror #3236: [weather] ignoreToday option is documented, but not implemented

d42ohpaz, 2023-10-17:
According to the [weather forecast options](https://docs.magicmirror.builders/modules/weather.html#weather-forecast-options), there is an `ignoreToday` option that would - as the name suggests - ignore the weather for the current day.

Grepping the module code base for `ignoreToday` returns zero results. Either the documentation should be updated to remove the option, or the option should be implemented. Personally, I'd prefer the option to be implemented. 

If it is as simple as editing the `forecast.njk` and starting the `currentStep` at `1` when `ignoreToday` is true, then I'd be happy to submit a PR. Otherwise, advice would be needed.

rejas, 2023-10-17:
The ignoreToday option is used here in the code: https://github.com/MichMich/MagicMirror/blob/c0a5f35a007c65b2744915d85404b482d2ef6eee/modules/default/weather/forecast.njk#L5

d42ohpaz, 2023-10-17:
@rejas My apologies; I mistakenly thought I was on the latest version of MagicMirror, but I am not. Thank you for pointing this out.

closed 2023-10-17 by d42ohpaz