MagicMirrorOrg/MagicMirror #3237: regular expression used as defaults value of a module, is corrupted.

eouia, 2023-10-18:
Recently, I found that when the regular expression is used as the default config value of a module, it would be corrupted.
I think it comes from configMerge or somewhere, but I cannot track exactly where the issue lies.

#### How to represent/prove
set some fields in `defaults` of MODULENAME.JS and `config` of `config.js`

```js
// In module file
Module.register("TEST", {
  defaults: {
    rex_default1: /ab+c/,
    rex_default2: [ 'ab+c', /ab+c/, new RegExp('ab+c') ],
    rex_config1: null,
    rex_config2: [],
...
```

```js
// in config.js
{
  module: 'TEST',
  config: {
    rex_config1: /ab+c/,
    rex_config2: [ 'ab+c', /ab+c/, new RegExp('ab+c') ],
...
//
```
Let's check which values are there.
```js
start: function () {
    console.log(this.config)
}
```

The Result;
[image]

eouia, 2023-10-18:
It doesn't matter.
General `Object.assign()` itself do it's role well.
[image]
I think `JSON.striingify()` somewhere makes this issue.

sdetweil, 2023-10-18:
is that the same data as the new Regex() which you reported in first message?

eouia, 2023-10-18:
??? sorry, I cannot catch your question.

In my original post, 
`rex_config1` and `rex_config2` were imported from 'config.js' and overrided as expected.
But,
`rex_default1` and `rex_default2` were corrupted. They should remain but changed as `/(?:)/`, and this value would definitely be derived from `new RegExp(false|null|undefined)`. So I guess somewhere `JSON.stringify` fails to convert proper string, so it might have `undefined` or `false` then tried to revamp RegExp Object from it.

sdetweil, 2023-10-18:
i would try to set the deepMerge: true option and see if that makes a difference.. it doesn't use Object.assign()

eouia, 2023-10-18:
Yes, I can avoid this problem by assigning regexp objects in code (usually `start()`) manually when the user configuration is not overriding them. It is just a little annoying and unexpected behaviour. That's all.

sdetweil, 2023-10-18:
the setting to add to your module before config is 
```
 configDeepMerge: true,
```

khassel, 2023-10-18:
the problem is [this line](https://github.com/MichMich/MagicMirror/blob/343e7de7bd295a69e68e4ee520552e0785a99e1c/js/module.js#L465) in `js/module.js`, before `cloneObject` the values are o.k., after the call they are corrupted.

the `cloneObject` function is defined in `js/class.js` [here](https://github.com/MichMich/MagicMirror/blob/343e7de7bd295a69e68e4ee520552e0785a99e1c/js/class.js#L88).

khassel, 2023-10-18:
we have already a working function to clone config objects, so replacing

```js
	const clonedDefinition = cloneObject(moduleDefinition);
```

with 

```js
	const clonedDefinition = configMerge(moduleDefinition);
```
 works.

Will do more tests and provide a PR tomorrow.

PR #3240 (2023-10-19): fix cloneObject() function to respect RegExp
fixes #3237
commit 0e2da630 (2023-10-19): fix cloneObject() function to respect RegExp (#3240)

fixes #3237
closed 2024-01-01 by MichMich