MagicMirrorOrg/MagicMirror #3239: [Weather] Show max and min of the day

Selyjohns, 2023-10-19:
Hello,

For the current day, the module display the maximum and minimum temperature.
But, the maximum isn't the maximum of the day, it is the current temperature in case of current temperature is lower than maximum temperature of the day.

I think this is the same with the minimum one.

Is it possible to fix the maximum and minimum value for current day, and not modify it to current temp ?

It will permit to show the current temp/the maximum of the day/the minimum of the day.
It is not very usefull to see the max temp as the same value as current temp.

Thank you so much for your great job !
Have a nice day.

khassel, 2023-10-19:
Please provide all info's requested in our issue template, additional:

- you are using the default weather module?
- if so, which provider?

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3/4, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 18 or later (recommended is 20).

**MagicMirror² Version**: Please let us know which version of MagicMirror² you are running. It can be found in the `package.json` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

Selyjohns, 2023-10-19:
> Please provide all info's requested in our issue template, additional:
>
> * you are using the default weather module?
> * if so, which provider?
>
> ## I found a bug in MagicMirror
> Please make sure to only [...]

Hello,

Yes it's the default weather module.
For exemple, for now my MM is displaying 
Current temp : 17,3

and some lines with today and next days
In the today line, i see 17,3 and  16,9

16,9 will the lowest temp tonight
17,3 is the current temp, and I think it will be better if the first temp displaying 25, the maximum reached today, instrad 17,3 that is current temp.

i will provide some information on my MM tomorrow, but all is up to date and no specific config 😊

Sorry if i'm not clear...

closed 2024-10-13 by github-actions[bot]