MagicMirrorOrg/MagicMirror #3245: Calendar Issue

Pollard-J, 2023-10-22:
Hi
I think you have recently updated the "calendarfetcher.js" file and now my calendar is blank as its not pulling the events through. I did have a backup of an old version but that doesnt work as it stops the magic mirror loading up.

sdetweil, 2023-10-22:
@Pollard-J correct... in 2.25 we removed the node_fetch and digest-fetch libraries and collapsed down to the node provided (abd supported) fetch function..   this was documented in the Contributions.md, and in the announcement
this is the reason node 18 or above is required, as that is where the nodejs fetch function was introduced. 

if you modified one of the MM provided files, you are responsible for its maintenance going forward

sdetweil, 2023-10-22:
yes, says it right there

fetch is not found..

SO,  you didn't upgrade to the required nodejs version..
```
node -v 
```
will tell you what you have

if you had used my upgrade script, I would have fixed this ..
see
https://github.com/sdetweil/MagicMirror_scripts

you can run this upgrade script again with force at the end instead of apply and it will fix the nodejs problem for you

Pollard-J, 2023-10-22:
Thank you for your help, much appreciated.

sdetweil, 2023-10-22:
@Pollard-J are you running now? can we close this?

Pollard-J, 2023-10-23:
I used your script and now it's updated and the node version is V20.8.0

But I still have no calendar events. I've deleted and re jnstalled the calendar module but still no change. Pm2 logs still say this...

[image]

sdetweil, 2023-10-23:
show me the ~/MagicMirror/installers/upgrade.log

Pollard-J, 2023-10-23:
Upgrade started - Mon Oct 23 09:09:25 BST 2023
system is Linux raspberrypi 5.15.32-v7l+ #1538 SMP Thu Mar 31 19:39:41 BST 2022 armv7l GNU/Linux
the os is Distributor ID: Raspbian Description: Raspbian GNU/Linux 11 (bullseye) Release: 11 Codename: bullseye

added 1 package, and audited 2 packages in 1s

found 0 vulnerabilities
  installing : node-v20.8.0
       mkdir : /usr/local/n/versions/node/20.8.0
       fetch : https://nodejs.org/dist/v20.8.0/node-v20.8.0-linux-armv7l.tar.xz
     copying : node/20.8.0
   installed : v20.8.0 (with npm 10.1.0)

Note: the node command changed location and the old location may be remembered in your current shell.
         old : /usr/bin/node
         new : /usr/local/bin/node
If "node --version" shows the old version then start a new shell, or reset the location hash with:
hash -r  (for bash, zsh, ash, dash, and ksh)
rehash   (for csh and tcsh)
doing test run = true, NO updates will be applied!
[96mCheck current Node installation ...[0m
[0mNode currently installed. Checking version number.
[0mMinimum Node version: [1mv20.8.0[0m
[0mInstalled Node version: [1mv20.8.0[0m
[92mNo Node.js upgrade necessary.[0m
[96mCheck current NPM installation ...[0m
[0mNPM currently installed. Checking version number.
[0mMinimum npm version: [1mV10.1.0[0m
[0mInstalled npm version: [1mV10.1.0[0m
[92mNo npm upgrade necessary.[0m

saving custom.css
remote name = origin
upgrading from version 2.25.0 to 2.25.0
fetching latest revisions
git fetch rc=0
current branch = master
On branch master
Your branch is up to date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	core
	css/save_custom.css

nothing added to commit but untracked files present (use "git add" to track)
no files different from github version
test merge result rc='' , if empty, no conflicts
skipping merge, only test run
Checking for modules with removed libraries
downloading dumpactivemodules script

processing dependency changes for active modules with package.json files
processing for module MMM-CalendarExt3 please wait
----------------------------------
skipped processing for MMM-CalendarExt3, doing test run
processing complete for module MMM-CalendarExt3
processing for module MMM-BackgroundSlideshow please wait
----------------------------------
skipped processing for MMM-BackgroundSlideshow, doing test run
processing complete for module MMM-BackgroundSlideshow
processing for module MMM-WeeklySchedule please wait
----------------------------------
skipped processing for MMM-WeeklySchedule, doing test run
processing complete for module MMM-WeeklySchedule
processing for module MMM-OpenWeatherForecast please wait
----------------------------------
skipped processing for MMM-OpenWeatherForecast, doing test run
processing complete for module MMM-OpenWeatherForecast
restoring custom.css
removing git alias
Upgrade ended - Mon Oct 23 09:10:08 BST 2023

sdetweil, 2023-10-23:
ok, good catch 

2 things

1 your WERE downlevel on nodejs.. now not (but theres a bug there too)
2.. you are doing the test run..   
     doing test run
     
so, take the SAME command and add 
```
 force
```
after it
(as your code is already on the latest MM release)

the bug is , I upgraded node, even tho were were only doing a test run... I shouldn't have made ANY changes...

Pollard-J, 2023-10-23:
That has fixed it. @sdetweil  Thank you for all your help.

closed 2023-10-23 by sdetweil