MagicMirrorOrg/MagicMirror #3250: Calendar events that have exclusion dates may not be excluded

jkriegshauser, 2023-10-26:
I have an event that is as follows:
```
BEGIN:VEVENT
DTSTART;TZID=America/Los_Angeles:20230315T181000
DTEND;TZID=America/Los_Angeles:20230315T195000
RRULE:FREQ=WEEKLY;UNTIL=20231129T075959Z;BYDAY=WE
EXDATE;TZID=America/Los_Angeles:20231025T181000
DTSTAMP:20231025T233434Z
UID:sdflbkasuhdb5fkauglkb@google.com
CREATED:20230306T193128Z
LAST-MODIFIED:20231024T222515Z
SEQUENCE:0
STATUS:CONFIRMED
SUMMARY:My Event
TRANSP:OPAQUE
END:VEVENT
```

Note that there is an EXDATE for today. However, it was showing on the Magic Mirror.

sdetweil, 2023-10-31:
note that the dates from rrule.between are jacked cause of the tz vs local thing..

so it got TODAYS date and the timezone time which makes the even tomorrow..(tz in the future of UTC) oops..

this seems to be because of the  rrule  byweekday   I have tried to recorrect those.. 

watch out for datekey being wrong, because date is damaged and not corrected 

	const dateKey = date.toISOString().substring(0, 10);

jkriegshauser, 2023-10-31:
@sdetweil it appears to be wrong even if the iCal event and local time are in the same timezone. In the event above, note `BYDAY=WE`, so it's on a Wednesday. When `rule.between()` is executed within the `America/Los_Angeles` timezone, the dates returned were exclusively on Tuesdays.

The node-ical library changes all `EXDATE` entries to ISO, so it is fairly easy to find those, assuming that the recurrence dates from `rrule` are correct (which they're not).

sdetweil, 2023-10-31:
assuming.. duh.. 

yes, you have to look at the dates returned to see that trhe date and time are mangled... date from one, tz from another.. 
when BYDAY/BYWEEKDAY.. if you remove that clause u get different date results..   

you can see it in the debug output

BUT as I said before WE cannot TELL its mangled with certainty,    as rrule is processing from the rrule object and not from the event object. 

I corrrect the non-excluded date/times when BYDAY is  set
```
// if the rrule byweekday WAS explicitly set , correct it
									// reduce the time by the offset
									if (curEvent.rrule.origOptions.byweekday !== undefined) {
										// Apply the correction to the date/time to get it UTC relative
										date = new Date(date.getTime() - Math.abs(24 * 60) * 60000);
									}
```

the 'problem' is that the tz adjustment sets the date to utc time,    1am UTC is yesterday 20:00 in ny.   and  today 11pm UTC is tomorrow 1am in german tz. and rrule expects local time.

jkriegshauser, 2023-10-31:
Do you know a better rrule js library to use? We could see about forking node-ical to use something else.

sdetweil, 2023-10-31:
I do not know of anything better

closed 2024-01-01 by khassel