MagicMirrorOrg/MagicMirror #3252: Feature request: Debug date

jkriegshauser, 2023-10-26:
I looked at documentation and through the code and didn't see any way to easily do this.

As a developer, it would be beneficial to have MagicMirror use a simulated debug date. This could be specified in *config.json* for example, and then become useful for tests.

This would essentially replace all calls to `moment()` with `moment(config.debugDate)` since `moment(undefined)` falls back to returning the current date/time. Anything else that doesn't use `moment()` would also need fixup.

buxxi, 2023-10-26:
Just add `Date.now = () => new Date('2001-02-03T04:05:06');` in your config.js

rejas, 2023-10-26:
> Just add `Date.now = () => new Date('2001-02-03T04:05:06');` in your config.js outside of the config-object

This might be a helpful addition to the module-development section of the documentation if someone wants to create a PR for that ;-)

jkriegshauser, 2023-10-26:
Setting `Date.now` to a lambda that responds with a specific date is an interesting idea, but it basically means that time stops. Sometimes we just want time to start at a specific time when MM is started 🤔

jkriegshauser, 2023-12-31:
> Just add `Date.now = () => new Date('2001-02-03T04:05:06');` in your config.js outside of the config-object

@buxxi this didn't quite work. The date/time typically in the top-middle shows the hard-coded date/time, but many modules still show actual time. Any other ideas?

jkriegshauser, 2023-12-31:
I went through and changed everywhere doing `new Date()` to `Date.now()` and that seemed to work. However, several external modules use `new Date()` and we should probably offer guidance on using `Date.now()` instead

jkriegshauser, 2023-12-31:
Also created https://github.com/MichMich/MagicMirror-Documentation/pull/218 for the documentation fixup

KristjanESPERANTO, 2023-12-31:
> However, several external modules use new Date() and we should probably offer guidance on using Date.now() instead

To see which modules are using `new Date()` I just added a check for this in my 3rd Party Module project - [here is the result](https://github.com/KristjanESPERANTO/MagicMirror-3rd-Party-Modules/blob/main/result.md). There are a few developers who have adapted their modules in the past according to this list. So this may help to decrease the use of `new Date()` in the future.

closed 2024-10-13 by github-actions[bot]