MagicMirrorOrg/MagicMirror #3253: Make User-Agent configurable

m-idler, 2023-10-26:
I just encountered an error regarding fetching a subscribed calendar. By debugging I could verify, that the issue is caused by the server, blocking requests based on matching `Node.js` in the `User-Agent` header. By reading through other issues i saw, that from time to time there have been some other issues regarding parts of the use agent (e.g. #2729 ). 
So I would suggest, to add a configuration option to optionally override the user-agent at least for default modules calendar and news (see https://github.com/MichMich/MagicMirror/blob/343e7de7bd295a69e68e4ee520552e0785a99e1c/modules/default/newsfeed/newsfeedfetcher.js#L105 and https://github.com/MichMich/MagicMirror/blob/343e7de7bd295a69e68e4ee520552e0785a99e1c/modules/default/calendar/calendarfetcher.js#L42 ). Other modules are then free to consume this option as well.

If you agree on that, i won't mind to open a PR to tackle it.

m-idler, 2023-10-30:
Temporarily fixed it by adding `sendheaders=User-Agent:....` as parameter to the url which is then parsed and applied here:
https://github.com/MichMich/MagicMirror/blob/master/js/server_functions.js#L72

As i didn't find anything in the documentation, would it make sense to add it somewhere?

sdetweil, 2025-08-25:
See https://forum.magicmirror.builders/topic/19849/calendar-fetch-error

KristjanESPERANTO, 2025-09-30:
Release 2.33.0 is out 🙂

PR #3255 (2025-08-27): Make `User-Agent` configurable
Fixes #3253 

Adds a configuration option to overwrite the default `User-Agent` header that is send at least by the calendar and news module. Allows other modules to use the individual user agent as well.

The configuration accepts either a string or a function:
```
var config =
	{
		...
		userAgent: 'Mozilla/5.0 (My User Agent)',
		...
	}
```
or
```
var config =
	{
		...
		userAgent: () => 'Mozilla/5.0 (My User Agent)',
		...
	}
```
commit 76da0aa5 (2025-08-27): Make `User-Agent` configurable (#3255)

Fixes #3253 

Adds a configuration option to overwrite the default `User-Agent` header
that is send at least by the calendar and news module. Allows other
modules to use the individual user agent as well.

The configuration accepts either a string or a function:
```
var config =
	{
		...
		userAgent: 'Mozilla/5.0 (My User Agent)',
		...
	}
```
or
```
var config =
	{
		...
		userAgent: () => 'Mozilla/5.0 (My User Agent)',
		...
	}
```

---------
closed 2025-09-30 by KristjanESPERANTO