MagicMirrorOrg/MagicMirror #3262: v2.25.0 - Clock module - (timezone: "Europe/Berlin") Returns Undefined

wbleek, 2023-11-04:
After updating to the newest version I get "undefined" with the clock module.

The clock module is the second module in my config.js script.
I only specify 

```
                {
                        module: "clock",
                        position: "top_left",
                        config: {
                                timezone: "Europe/Berlin",
                                showWeek: true
                        }
                },

```
This might be related to issue 3143 https://github.com/MichMich/MagicMirror/issues/3143

sdetweil, 2023-11-04:
I don't see that , for either of these  on 2.25 or on 2.26-develop
```
		},
                {
                        module: "clock",
                        position: "top_left",
                        config: {
                                timezone: "Europe/Berlin",
                                showWeek: true
                        }
                },
		{
			module: "clock",
			disabled: true,
			position: "top_left",
			config:{
				displaySeconds:false,
				timezone:"Europe/Berlin"
			}
		},
```

khassel, 2023-11-04:
I can't reproduce this. Please provide a complete `config.js` example where the error occurs, thanks.

khassel, 2024-01-03:
Closing this because the required information was not provided.

closed 2024-01-03 by khassel