MagicMirrorOrg/MagicMirror #3263: RSS Reader Api Support in newsfeed module

b-reich, 2023-11-05:
It would be a cool feature if the `newsfeed` module had support for rss reader api like [fever](https://freshrss.github.io/FreshRSS/en/developers/06_Fever_API.html) or [greader](https://freshrss.github.io/FreshRSS/en/developers/06_GoogleReader_API.html). Both apis are often supported by rss feeder

I think it is practical to maintain your rss feeds in only one place. it may be practical to display only tagged feeds on the mirror.

rejas, 2023-11-05:
Do you have any example urls that one can use to test the current newsfeed module with? So we might see what has to be done to get those supported? 
Also: For me newb, what does fever and greader offer for adavantages?

b-reich, 2023-11-05:
Think you can test it with freshrss demo feed https://demo.freshrss.org/api/ (when not, i could provide you an account on my instance)
there are some api docs linked here [fever](https://freshrss.github.io/FreshRSS/en/developers/06_Fever_API.html) or [greader](https://freshrss.github.io/FreshRSS/en/developers/06_GoogleReader_API.html)

I think the greader one is bit more common because Google Reader was so popular. Maybe @Alkarex (Maintainer of fresshrss) can provide some infos

Alkarex, 2023-11-05:
Let me know if you need anything concrete. The GReader API is much more comprehensive and robust than the Fever API

rejas, 2023-11-18:
I personally dont have the time to do this. If somebody wants to take this up, feel free to comment here.

bugsounet, 2023-11-18:
I will take a look what can I do (when I have some time) ;)

Alkarex, 2023-11-19:
I anyone starts an implementation, please check https://freshrss.github.io/FreshRSS/en/developers/06_GoogleReader_API.html
and in particular:

* [*Synchronisation recommendation* by Alkarex](https://github.com/FreshRSS/FreshRSS/issues/2566#issuecomment-541317776)
* [*The Right Way to Sync* by BazQux](https://github.com/bazqux/bazqux-api#user-content-the-right-way-to-sync)