MagicMirrorOrg/MagicMirror #3266: CORS proxy fails for images

ToddRK, 2023-11-10:
I modified the default newsfeed module to return images. This works fine when the CORS proxy isn't being used but I get broken link icons in place of the images when the CORS proxy is used, i.e. when useCorsProxy is true in config.js and the images are accessed via a url such as http://localhost:8080/cors?url=https://globalnews.ca/wp-content/uploads/2023/11/icbc-pic.jpg.

Most of the news feed images I’ve been accessing are in webp format even though their URLs suggest they are JPGs.

I'm not sure what the problem is but I noticed the CORS proxy seems to be setting the Content-Type and Content-Length wrong.

When the CORS proxy is off, the Content-Type is correctly set to “image/webp” and the Content-Length is correct.
When the CORS proxy is enabled, the Content-Type is incorrectly set to “image/jpeg” and the Content-Length is much larger than it should be.

I tried adding expectedheaders=Content-Type,Content-Length to the CORS request but that didn’t seem to affect anything.

Here are headers for the same newsfeed image without the CORS proxy enabled, i.e. image retrieval works, and with it, i.e. fails.

Without CORS proxy:

Request
---------
Request URL: https://globalnews.ca/wp-content/uploads/2023/11/icbc-pic.jpg?quality=85&strip=all
Request Method: GET
Status Code: 200
Remote Address: 192.0.66.184:443 (???)
Referrer Policy: no-referrer

Request headers
------------------
:authority: [globalnews.ca](http://globalnews.ca/)
:method: GET
:path: /wp-content/uploads/2023/11/icbc-pic.jpg?quality=85&strip=all
:scheme: https
Accept: image/avif,image/webp,image/apng,image/svg+xml,image/,/*;q=0.8
Accept-Encoding: gzip, deflate, br
Accept-Language: en-GB
Sec-Ch-Ua: “Not)A;Brand”;v=“24”, “Chromium”;v=“116”
Sec-Ch-Ua-Mobile: ?0
Sec-Ch-Ua-Platform: “Linux”
Sec-Fetch-Dest: image
Sec-Fetch-Mode: no-cors
Sec-Fetch-Site: cross-site
User-Agent: Mozilla/5.0 (X11; Linux armv7l) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/116.0.5845.190 Electron/26.2.4 Safari/537.36

Response headers
--------------------
Accept-Ranges: bytes
Cache-Control: max-age=31536000
Content-Length: 84572
Content-Type: image/webp
Date: Mon, 06 Nov 2023 04:46:32 GMT
Etag: “b8bb49c46d684e95”
Last-Modified: Mon, 06 Nov 2023 01:55:46 GMT
Server: nginx
Strict-Transport-Security: max-age=86400
Vary: Accept
X-Cache: HIT
X-Rq: sea4 116 203 443

When the CORS proxy is used, the image isn’t returned and a broken link icon is displayed.

Request
---------
Request URL: http://localhost:8080/cors?url=https://globalnews.ca/wp-content/uploads/2023/11/icbc-pic.jpg?quality=85&strip=all
Request Method: GET
Status Code: 200 OK
Remote Address: [::1]:8080
Referrer Policy: no-referrer

Request headers
------------------
Accept: image/avif,image/webp,image/apng,image/svg+xml,image/,/*;q=0.8
Accept-Encoding: gzip, deflate, br
Accept-Language: en-GB
Connection: keep-alive
Host: localhost:8080
Sec-Ch-Ua: “Not)A;Brand”;v=“24”, “Chromium”;v=“116”
Sec-Ch-Ua-Mobile: ?0
Sec-Ch-Ua-Platform: “Linux”
Sec-Fetch-Dest: image
Sec-Fetch-Mode: no-cors
Sec-Fetch-Site: same-origin
User-Agent: Mozilla/5.0 (X11; Linux armv7l) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/116.0.5845.190 Electron/26.2.4 Safari/537.36

Response headers
--------------------
Access-Control-Allow-Origin: *
Connection: keep-alive
Content-Length: 274934
Content-Type: image/jpeg; charset=utf-8
Date: Mon, 06 Nov 2023 04:41:38 GMT
Etag: W/“431f6-FDM8wfRxenBocIzJBv0QxLtyTa0”
Keep-Alive: timeout=5
Referrer-Policy: no-referrer
Strict-Transport-Security: max-age=15552000; includeSubDomains
X-Content-Type-Options: nosniff
X-Dns-Prefetch-Control: off
X-Download-Options: noopen
X-Frame-Options: SAMEORIGIN
X-Permitted-Cross-Domain-Policies: none
X-Xss-Protection: 0

ToddRK, 2023-11-10:
I forgot to mention I'm using MM 2.25.0

khassel, 2023-11-25:
I did some tests with curl using the direct url and the cors-url of mm.

It seems this is a problem of the webp format, jpeg ist working with the cors proxy.

The cors proxy was introduced as a hack in this project to avoid cors errors in the browser and is needed for some urls in the default newsfeed and weather module. It was never meant as a full cors proxy which covers all use cases.

If you find a solution for this we are happy to accept a PR but I think this has a very low priority here and I have not enough knowledge to fix this.