MagicMirrorOrg/MagicMirror #3267: Weird broadcasted form of multi symbols of calendar module.

eouia, 2023-11-12:
I remember it was not so,  but multiple symbols are broadcasted weirdly at some point.

When I set some symbols in a calendar like  this;
```js
symbol: ["file", "calendar"],
```

The expected broadcasted data(And the historical format what I remember) would be like this;
```js
symbol: ["fa fa-fw fa-file", "fa fa-fw fa-calendar"],
```

Bur really broadcasted would be;
```js
symbol: ["fa fa-fw fa-file, fa fa-fw fa-calendar"],
```

Is this intentionally designed so? Or just an unexpected accident?

rejas, 2023-11-18:
Thanks for the report (and sorry for the late replay). Can you pinpoint when this behaviour changed?

sdetweil, 2024-10-11:
I am fixing things in calendar now.. @eouia  can u answer @rejas

sdetweil, 2024-10-17:
I see the bug, checking on how to fix

sdetweil, 2024-10-21:
so, here is my proposed change 
```
	getCalendarPropertyAsArray (url, property, defaultValue) {
		let p = this.getCalendarProperty(url, property, defaultValue);
		if (property === "symbol" || property === "recurringSymbol" || property === "fullDaySymbol") {
			const className = this.getCalendarProperty(url, "symbolClassName", this.config.defaultSymbolClassName);
			if(p instanceof Array)       //  add
				p.push(className) // add
			else   // add
				p = className + p;  // old code assumed property was string
		}

		if (!(p instanceof Array)) p = [p];
		return p;
	},
```

commit 5232f46d (2024-12-08): fix #3267 again this time, dropped from big cal update (#3650)

this change was dropped from #3267 by mistake
PR #3681 (2025-01-05): fix #3267, CORRECTLY, add testcase, add testcase for #3279
fixes #3267 AGAIN, correctly, add testcase
add testcase for #3679 , broadcast clipping incorrectly

I added a test module (tests/testNotification) to catch the notification and check the count. (one  way to configure)
i put this module in the tests folder, and added /tests to the server paths. 
(can't have a module in a nested folder, like tests/modules/xxx)

but I have a problem.  i can run the test config (MM_CONFIG_FILE), and the two modules work correctly, 
but in the spec runner, the calendar module times out on the broadcast test [... cut]
commit 5e337f8b (2025-01-05): fix #3267, CORRECTLY, add testcase, add testcase for #3279 (#3681)

fixes #3267 AGAIN, correctly, add testcase
add testcase for #3679 , broadcast clipping incorrectly

I added a test module (tests/testNotification) to catch the notification
and check the count. (one way to configure)
i put this module in the tests folder, and added /tests to the server
paths.
(can't have a module in a nested folder, like tests/modules/xxx)

but I have a problem. i can run the test config (MM_CONFIG_FILE), and
the two modules work correctly,
but in the spec runner, the calendar module times out on the broadcast [... cut]
closed 2025-01-01 by sdetweil