MagicMirrorOrg/MagicMirror #3271: Can't get any module to display Nextcloud Tasks/Deck items.

bigrjsuto, 2023-11-20:
I can't seem to get any of these three modules available to load items from my Nextcloud:

* [MMM-iCal-ToDo](https://github.com/nixnuex/MMM-iCal-ToDo) (Left)
   * I've tried every combination of link I could think of.
   * Maybe I need to use a different auth?
* [MMM_ncdeck](https://github.com/motey/MMM_ncdeck) (Middle)
   * I put the correct board # 
* [NextCloud-Tasks](https://github.com/starlingfire/MMM-NextCloud-Tasks) (Right)
   * I'm using the private link and App Password as instructed.

Can someone share a working config?
[image]

bigrjsuto, 2023-11-20:
{
                        module: 'MMM-NextCloud-Tasks',
                        position: "bottom_right",
                                config: {
                                        updateInterval: 60000,
                                        listUrl: "https://cloud.xxxxxxx.com/remote.php/dav/calendars/xxxxxxx/personal/",
                                        hideCompletedTasks: true,
                                        sortMethod: "priority",
                                        colorize: true,
                                        webDavAuth: {
                                                username: "xxxxxxx",
                                                password: "NCAppPassword",
                                                }
                                        }
                },
                {
                        module: "MMM_ncdeck",
                        position: "bottom_center",
                        config: {
                        nextcloud: {
                          host: "https://cloud.xxxxxxx.com/index.php/apps/deck/#/",
                          user: "xxxxxxx",
                          pass: "ncpassword",
                        },
       {
                        module: "MMM-iCal-ToDo",
                        position: "bottom_left",        // This can be any of the regions. Best results in left or right regions.
                        config: {
                                header: 'Tasks for Rob',
                                colored: true,
                                coloredSymbolOnly: false,
                                todos: [
                                        {
                                        url: 'https://cloud.xxxxxx.com/remote.php/dav/calendars/xxxxxxx/personal/',
                                        symbol: 'check',
                                        auth: {
                                            user: 'xxxxxxx',
                                            pass: 'ncpassword',
                                            method: 'basic'
                                                }
                                        },
                                        ],
                                },
                },

sdetweil, 2023-11-20:
we don't provide support 3rd party modules here on this project.

you can post in the GitHub repo for each to engage the module author, or try the forums for other users assistance.

closed 2023-11-21 by bigrjsuto