MagicMirrorOrg/MagicMirror #3273: Openmeteo shows apparent temperature in hourly and daily

F3RIXX, 2023-11-23:
I use openmeteo as the weatherprovider.
It works good.

In the current weather is all fine.

But when I put the hourly and the daily in, it shows me the apparent temperature and not the temperature itself.

		{
			module: "weather",
			position: "top_right",
			config: {
				weatherProvider: "openmeteo",
				apiBase: "https://api.open-meteo.com/v1",
				lat: "xx",
				lon: "xx",
				windUnits: "kmh",
				updateInterval: "300000",
				appendLocationNameToHeader: false,
				showSun: false,
				showHumidity: "true"
			}
		},
		{
			module: "weather",
			position: "top_right",
			config: {
				weatherProvider: "openmeteo",
				apiBase: "https://api.open-meteo.com/v1",
				lat: "xx",
				lon: "xx",
				type: "hourly",
				maxEntries: "7",
				showPrecipitationAmount: "true",
				showPrecipitationProbability: "true",
				updateInterval: "1800000",
				appendLocationNameToHeader: false,
				fade: false
			}
		},
		{
			module: "weather",
			position: "top_right",
			config: {
				weatherProvider: "openmeteo",
				apiBase: "https://api.open-meteo.com/v1",
				lat: "xx",
				lon: "xx",
				maxNumberOfDays: "7",
				type: "daily",
				showPrecipitationAmount: "true",
				showPrecipitationProbability: "true",
				updateInterval: "7200000",
				appendLocationNameToHeader: false,
				fade: false
			}
		},

I'm confused, and don't know where I have to look to change it.
I looked in the provider folder the openmeteo.js too, and changed there something from apparent to temperature_2m, but it didn't worked.

Can you/somebody help me?

Thanks

F3RIXX, 2024-04-29:
I finally found it!
(To be honest, I didn't looked a long time after it, because no time, and then my whole MM didn't worked anymore, so I had to install it completly new. But because of this, I found the solution):

Under MagicMirror\modules\default\weather\providers\openmeteo.js is this part (lowerd third):

	// Implement WeatherDaily generator.
	generateWeatherObjectsFromForecast (weathers) {
		const days = [];

		weathers.daily.forEach((weather) => {
			const currentWeather = new WeatherObject();

			currentWeather.date = weather.time;
			currentWeather.windSpeed = weather.windspeed_10m_max;
			currentWeather.windFromDirection = weather.winddirection_10m_dominant;
			currentWeather.sunrise = weather.sunrise;
			currentWeather.sunset = weather.sunset;
			currentWeather.temperature = parseFloat((weather.apparent_temperature_max + weather.apparent_temperature_min) / 2);
			currentWeather.minTemperature = parseFloat(weather.apparent_temperature_min);
			currentWeather.maxTemperature = parseFloat(weather.apparent_temperature_max);
			currentWeather.weatherType = this.convertWeatherType(weather.weathercode, currentWeather.isDayTime());
			currentWeather.rain = parseFloat(weather.rain_sum);
			currentWeather.snow = parseFloat(weather.snowfall_sum * 10);
			currentWeather.precipitationAmount = parseFloat(weather.precipitation_sum);
			currentWeather.precipitationProbability = parseFloat(weather.precipitation_probability);
			currentWeather.uv_index = parseFloat(weather.uv_index_max);

			days.push(currentWeather);
		});

		return days;
	},

	// Implement WeatherHourly generator.
	generateWeatherObjectsFromHourly (weathers) {
		const hours = [];
		const now = moment();

		weathers.hourly.forEach((weather, i) => {
			if ((hours.length === 0 && weather.time.hour() <= now.hour()) || hours.length >= this.config.maxEntries) {
				return;
			}

			const currentWeather = new WeatherObject();
			const h = Math.ceil((i + 1) / 24) - 1;

			currentWeather.date = weather.time;
			currentWeather.windSpeed = weather.windspeed_10m;
			currentWeather.windFromDirection = weather.winddirection_10m;
			currentWeather.sunrise = weathers.daily[h].sunrise;
			currentWeather.sunset = weathers.daily[h].sunset;
			currentWeather.temperature = parseFloat(weather.apparent_temperature);
			currentWeather.minTemperature = parseFloat(weathers.daily[h].apparent_temperature_min);
			currentWeather.maxTemperature = parseFloat(weathers.daily[h].apparent_temperature_max);
			currentWeather.weatherType = this.convertWeatherType(weather.weathercode, currentWeather.isDayTime());
			currentWeather.humidity = parseFloat(weather.relativehumidity_2m);
			currentWeather.rain = parseFloat(weather.rain);
			currentWeather.snow = parseFloat(weather.snowfall * 10);
			currentWeather.precipitationAmount = parseFloat(weather.precipitation);
			currentWeather.precipitationProbability = parseFloat(weather.precipitation_probability);
			currentWeather.uv_index = parseFloat(weather.uv_index);

			hours.push(currentWeather);
		});

		return hours;
	},

And under temperature, minTemperature and maxTemperature it says the apparent_temperature for daily and hourly. Swap that with temperature_2m everywhere and you have the real temperature.

Oh, and for the Precipitation Probability for daily it has to be precipitation_probability_max (Maybe openmeteo changed this).

closed 2024-04-29 by F3RIXX

rejas, 2024-04-29:
Shouldnt this then get fixed with a PR or didnt I understand your answer correctly?

F3RIXX, 2024-04-29:
What's a PR? :O

rejas, 2024-04-29:
Oh, sorry. What I wanted to ask was: You found out that the MagicMirror Code doesnt handle the OpenMeteo data correctly? And it wasnt an error in your config?
A PR would be a "PullRquest" which contains code changes to fix the problems in our code.

F3RIXX, 2024-04-29:
No problem ;)
Well, in the end the data was used correctly, just not the way I wanted it. (I wanted the real temperature, displayed was felt like). And I tried various things in the config, but nothing worked. And now I just came across openmeteo.js.

Ahh, okayyy.
Unfortunately, I can't say how to deal with this now. I haven't been on GitHub myself for very long, so I've now written down the solution so that someone can find it if they search for it. I don't know whether you should/have to adapt the code directly.