MagicMirrorOrg/MagicMirror #3282: How to build a magic mirror

jkatsnelson, 2023-12-02:
I bet a lot of people coming to your site for the first time are wondering "how do I build a magic mirror?" And all they will find is software documentation. I'd recommend just putting a link to your blog post or something near the top to send people towards the end product so they can understand the big picture.

sdetweil, 2023-12-02:
I'd send them to the forums, 
https://forum.magicmirror.builders/category/12/show-your-mirror

there are thousands of ways. 
I don't use a mirrored surface.. don't NEED another 'mirror' in the house.. but the info is what I want

khassel, 2023-12-13:
> I'd recommend just putting a link to your blog post or something near the top

Do you mean the start site of this github repo or the [documentation](https://docs.magicmirror.builders/) or ...?

jkatsnelson, 2023-12-13:
Either one - just hit the link and imagine you've come to these pages for the first time, because you heard about this concept of magic mirrors and have no idea where to start 😀