MagicMirrorOrg/MagicMirror #3283: Electron Timezone set to UTC, not system timezone

kjendal, 2023-12-03:
After upgrading to the latest MM, all modules which do not support explicit timezone overrides are displaying time in UTC (calendar, weather, clock) on the native display (a browser-based view displays in the browser's timezone).  The "fix" has been reported here: (https://www.reddit.com/r/MagicMirror/comments/169w63z/comment/k3tfd8m/?utm_source=reddit&utm_medium=web2x&context=3)

Synopsis: add a line to js.electron.js to force the timezone: 
process.env.TZ = 'America/New_York';

sdetweil, 2023-12-03:
also, no need to change code, just
export TZ= in the script that starts mm

sdetweil, 2023-12-04:
I don't see that behavior on any system. old buster, bullseye, or bookworm

and so far, your issue outstanding, every time it's been a misconfigured system timezone.

the fact that one can override it indicates system setting issue.

closed 2024-01-18 by khassel