MagicMirrorOrg/MagicMirror #3288: Request: List of environment variables for documentation purposes

BKeyport, 2023-12-11:
The recent issue with misquoting how to use an environment variable makes me curious - what variables are available, and how are they used? I'd like to develop out documentation on how to use them, etc. I've been thinking of working on the documentation somewhat, and this seems like a good place to start.

sdetweil, 2023-12-11:
there are only 2
MM_CONFIG_FILE
and
MM_PORT

there is also some new support for templating, this IS documented.  
this basically provides a way to take the apikey, userid  passwords and separate them from the config file..  
using the envsub  replacement support.

khassel, 2023-12-13:
If someone is willing to update the documentation concerning the above two MM_ env vars I can move this issue into the documentation repo.

BKeyport, 2023-12-13:
I’ve been thinking of rewriting the config section of the docs due to lots
of issues in the forums. Move it over and I’ll start working on it.