MagicMirrorOrg/MagicMirror #3291: one calendar test fails since today

khassel, 2023-12-13:
```bash
  ● Calendar module › exdate check › should show the recurring event 51 times (excluded once) in a 364-day (inclusive) period

    expect(received).resolves.toBe()

    Received promise rejected instead of resolved
    Rejected to value: [Error: expect(received).toHaveLength(expected)·
    Expected length: 51
    Received length: 50
    Received object: [<tr class="event-wrapper normal event"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">My Event</td><td class="time light ">Dec 20th</td></tr>, <tr class="event-wrapper normal event"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">My Event</td><td class="time light ">Dec 27th</td></tr>, <tr class="event-wrapper normal event"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">My Event</td><td class="time light ">Jan 3rd</td></tr>, <tr class="event-wrapper normal event"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">My Event</td><td class="time light ">Jan 10th</td></tr>, <tr class="event-wrapper normal event"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">My Event</td><td class="time light ">Jan 17th</td></tr>, <tr class="event-wrapper normal event"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">My Event</td><td class="time light ">Jan 24th</td></tr>, <tr class="event-wrapper normal event"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">My Event</td><td class="time light ">Jan 31st</td></tr>, <tr class="event-wrapper normal event"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">My Event</td><td class="time light ">Feb 7th</td></tr>, <tr class="event-wrapper normal event"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">My Event</td><td class="time light ">Feb 14th</td></tr>, <tr class="event-wrapper normal event"><td class="symbol align-right "><span class="fas fa-fw fa-calendar-alt" /></td><td class="title bright ">My Event</td><td class="time light ">Feb 21st</td></tr>, …]]

       95 |
       96 |             it("should show the recurring event 51 times (excluded once) in a 364-day (inclusive) period", async () => {
    >  97 |                     await expect(testElementLength(".calendar .event", 51)).resolves.toBe(true);
          |                           ^
       98 |             });
       99 |     });
      100 |

      at expect (node_modules/expect/build/index.js:113:15)
      at Object.expect (tests/e2e/modules/calendar_spec.js:97:10)
```

I was able to fake the date in my test container:
- running with a date before today: test succeeds
- running with a date of today or later: test fails

I think this has something to do with the data used in `tests/mocks/calendar_test_exdate.ics`.

So I think we should remove the test until someone has an idea to fix it.

rejas, 2023-12-14:
maybe the original author @jkriegshauser can help us out witrh a fix?

jkriegshauser, 2023-12-15:
@rejas I'll try to take a look when i get some time. I've noticed issues since Los_Angeles timezone went off DST. I still highly suspect the `rrule` js module since there are hacks around making it work.

khassel, 2023-12-15:
Digging into this it seems the test still depends on the timezone used.

With setting the timezone to UTC before running the tests it seems to work. Will provide a PR and we will see if this fix is sufficent.

khassel, 2024-01-03:
this test still fails sometimes, seems to be time dependent

```bash
FAIL tests/electron/modules/calendar_spec.js (12.048 s)
  ● Calendar module › Exdate check › should show the recurring event 51 times (excluded once) in a 364-day (inclusive) period

    expect(received).toBe(expected) // Object.is equality

    Expected: 51
    Received: 50

      55 | 			expect(elem).not.toBeNull();
      56 | 			const cnt = await loc.count();
    > 57 | 			expect(cnt).toBe(51);
         | 			            ^
      58 | 		});
      59 | 	});
      60 | });

      at Object.toBe (tests/electron/modules/calendar_spec.js:57:16)
```

jkriegshauser, 2024-01-04:
Yep, something is still funky in the latest release. I had an event show up on MM today that was deleted (in `exdate`). I'll look into it

jc43081, 2024-01-07:
I am not sure this is related, but it would seem to fit the failure. I am seeing two different recurring meetings where I have events listed for all future dates, but not the original date. For example, I have an event set to recur every 2nd Wednesday starting January 10th. I don't see it on the 10th, but I do see it on all subsequent 2nd Wednesdays through the rest of the year.

jkriegshauser, 2024-01-07:
I think I know what it is. I’m looking into it.

PR #3293 (2023-12-17): fix calendar test exdate check
fixes #3291
commit a7af76b6 (2023-12-17): fix calendar test exdate check (#3293)

fixes #3291
closed 2024-03-27 by khassel