MagicMirrorOrg/MagicMirror #3292: Weather module (yr) hung on 'Loading ...'

basti0001, 2023-12-15:
First of all there is an Bug #3227 already but I suppose it's not the same issue. 

**System Information:**
rasspi, debian bookworm
MM2 version: /tag/v2.25.0
node -v: v18.17.1
npm --version: 6.13.4

**Logs:**
Loading with default config.js.sample as config.js I observe 2 instances of weather module stuck in loading. Here the log entries:
0|mm2      | npm WARN npm npm does not support Node.js v18.17.1
0|mm2      | npm WARN npm You should probably upgrade to a newer version of node as we
0|mm2      | npm WARN npm can't make any promises that npm will work with this version.
0|mm2      | npm WARN npm Supported releases of Node.js are the latest release of 6, 8, 9, 10, 11, 12, 13.
0|mm2      | npm WARN npm You can find the latest version at https://nodejs.org/
0|mm2      | [13303:1215/095553.718257:ERROR:bus.cc(399)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
0|mm2      | [13303:1215/095553.718786:ERROR:bus.cc(399)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
0|mm2      | [13303:1215/095554.222237:ERROR:bus.cc(399)] Failed to connect to the bus: Could not parse server address: Unknown address type (examples of valid types are "tcp" and on UNIX "unix")
0|mm2      | [13338:1215/095556.961718:ERROR:viz_main_impl.cc(186)] Exiting GPU process due to errors during initialization
0|mm2      | [13347:1215/095557.960684:ERROR:command_buffer_proxy_impl.cc(128)] ContextResult::kTransientFailure: Failed to send GpuControl.CreateCommandBuffer.

/home/pi/.pm2/logs/mm2-out-0.log last 10 lines:
0|mm2      | [15.12.2023 09:55.53.796] [LOG]   Starting node helper for: calendar
0|mm2      | [15.12.2023 09:55.53.798] [LOG]   Connecting socket for: newsfeed
0|mm2      | [15.12.2023 09:55.53.801] [LOG]   Starting node helper for: newsfeed
0|mm2      | [15.12.2023 09:55.53.804] [LOG]   Sockets connected & modules started ...
0|mm2      | [15.12.2023 09:55.53.938] [LOG]   Launching application.
0|mm2      | [15.12.2023 09:56.03.075] [LOG]   Create new calendarfetcher for url: https://ics.calendarlabs.com/76/mm3137/US_Holidays.ics - Interval: 3600000
0|mm2      | [15.12.2023 09:56.03.284] [LOG]   Create new newsfetcher for url: https://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml - Interval: 300000
0|mm2      | [15.12.2023 09:56.03.295] [INFO]  Checking git for module: MagicMirror
0|mm2      | [15.12.2023 09:56.04.399] [INFO]  Calendar-Fetcher: Broadcasting 13 events.
0|mm2      | [15.12.2023 09:56.04.428] [INFO]  Newsfeed-Fetcher: Broadcasting 15 items.

I also obtain this behavior when I switch to the default operation mm2 configuration file. But it's reproducable with config default. 
Yes I see npm warnings, that occur. I installed on debian node18 from repro, however there are warnings concerning npm version. I don't know exactly if they are the root for this behavior. 

What I also obtain is: switching back to the version, where there was the older module design of the weathermodule, everything works fine (with operating config.js as well as config.js.sample)
In my case I can confirm, that weather module is not working in my case for
v2.25.0
v2.24.0
...
I guess it's strange since I'm for shure not the only one using this module. So I think maybe I'm doing something wrong. 

How shall we proceed? I will try to dig in and debug out some Log.Info("") dumps to get an idea. Any better ideas?

khassel, 2023-12-15:
show your `config.js` (but without secrets, replace them with e.g. xxx )

> Loading with default config.js.sample as config.js I observe 2 instances of weather module stuck in loading.

this is normal behavior as long as you don't replace `apiKey: "YOUR_OPENWEATHER_API_KEY"` with a real key

basti0001, 2023-12-17:
Hi khassel,
ty for your help and answer. The API key was correct in my config.js. 
I went step by step throught all configs. 
I missed the following public config vars:

  useHttps: false,              // Support HTTPS or not, default "false" will use HTTP
  httpsPrivateKey: "",  // HTTPS private key path, only require when useHttps is true
  httpsCertificate: "",         // HTTPS Certificate path, only require when useHttps is true

The city ID was correct. The appKey was also correct. 

By merging the v2.25.0 config.sample into my operating config.js, I was successfull. Module starts as expexted. 

So all in all I guess it's like presumed not a real bug, the problem was i think the missing global config vars, that have not been part of my config in the first place I opened the issue. 

Ty for your time and help, I think we can close this issue. Sry I wasted your time, by not merging new config files into my default operating one. Will not happen again. 

Cheers

Sebastian

closed 2023-12-17 by rejas