MagicMirrorOrg/MagicMirror #3298: Custom Visual configuration

martinljor, 2023-12-21:
Hi Everyone,

Just to share my configuration running MM on Raspberry Pi 3+ with the following modules:

*Clock
*Weather
*MMM-CalendarExt2
*MMM-Unsplash

Some changes on CSS to:
*Increase month label x3
*hide week numbers

My configuration for config.js:
#############
let config = {
        address: "localhost",   // Address to listen on, can be:
                                                        // - "localhost", "127.0.0.1", "::1" to listen on loopback interface
                                                        // - another specific IPv4/6 to listen on a specific interface
                                                        // - "0.0.0.0", "::" to listen on any interface
                                                        // Default, when address config is left out or empty, is "localhost"
        port: 8080,
        basePath: "/",                  // The URL path where MagicMirror² is hosted. If you are using a Reverse proxy
                                                        // you must set the sub path here. basePath must end with a /
        ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1"],   

        useHttps: false,                // Support HTTPS or not, default "false" will use HTTP
        httpsPrivateKey: "",    // HTTPS private key path, only require when useHttps is true
        httpsCertificate: "",   // HTTPS Certificate path, only require when useHttps is true

        language: "es",
        locale: "es-AR",
        logLevel: ["INFO", "LOG", "WARN", "ERROR"], // Add "DEBUG" for even more logging
        timeFormat: 24,
        units: "metric",

        modules: [
    {
        module: 'MMM-Unsplash',
        position: 'fullscreen_below',
        config: {
       collections: 'myCollection',
                apiKey: 'myAPI'
        },
    },

                {
                        module: "clock",
                        position: "top_left",
                        config: {
                                displaySeconds: false,
                                dateFormat: "dddd, D MMM",
                                },
                },
                {
                module: 'MMM-CalendarExt2',
                        config: {
                        scenes:[
                        {
                                name: "Familia",
                                views: ["Calendar View"],
                        }
                        ],
                        views:[
                        {
                                name: "Calendar View",
                                mode: "month",
                                slotMaxHeight: "250px",
                                weeksFormat: "w",
                                position: "middle_center",
                                fromNow: 0,
                                slotCount: 1,
                                locale: "es-AR",
                                showEnd: "true",
                                dateFormat: "d",
                                useEventTimeRelative: "false",
                                hideOverflow: "false",
                                slotMaxHeight: "auto",
                                slotTitleFormat: "D",
                                slotSubTitleFormat: "dd",
filterPassedEvent: "false",
                                calendars: ["Familia"]
                        },
                        ],
                calendars: [
                        {
                                name: "Familia",
                                url: "myLink"
                                className: "Familia"
                        },
                ],
        },
    },

                {
                        module: "weather",
                        position: "top_right",
                        header: "",
                        config: {
                                weatherProvider: "openweathermap",
                                type: "current",
                                units: "metric",
                                showPrecipitationProbability: "false",
                                showPeriod: "false",
                                onlyTemp: "true",
                                showFeelsLike: "false",
                        //      location: "Buenos Aires",
                                locationID: "3433955", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
                                apiKey: "myAPI"
                        }
                },
        ]
};
/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}

#############

Any suggestion is welcome :)

KristjanESPERANTO, 2023-12-21:
Hello @martinljor 🙂

There is an extra section in the forum for this: https://forum.magicmirror.builders/category/12/show-your-mirror.

And if possible, add a photo or a screenshot.

khassel, 2023-12-21:
As already mentioned by @KristjanESPERANTO this is not the right place for showing mirrors or configs, please use the forum for this, thanks!

closed 2023-12-21 by khassel