MagicMirrorOrg/MagicMirror #3302: Environment Variables disable features....

BKeyport, 2023-12-23:
In researching for documentation purposes I've been finding all sorts of issues using the environment variables to multi-screen. I'm using bookworm, wayland turned off. 

1) using MM_CONFIG_FILE produces weird results: 
     A) the following config options are ignored: port, electronOptions, address (see next) 
     B) It'll set up the address on [::1]:8080 by default, rather than whatever address is commanded to do. I don't use IPV6 due to ISP non-support, so that puts it unavailable. 
     C) Due to this, was unable to test if ipWhitelist functions, but all other seems to function. 

2) using MM_PORT does move the port to the correct port, but still leaves it on address [::1]

relevant config2.js portion
```js
var config = {
	address: "0.0.0.0",
	port: 8081, 
	ipWhitelist: [], 
	language: "en",
	timeFormat: 12,
	units: "imperial",
	//logLevel: ["INFO", "LOG", "WARN", "ERROR", "DEBUG"],
	modules: [
```
bash script launch (mm2.sh): 
```bash
cd /home/bkey1970/MagicMirror
export MM_CONFIG_FILE=${pwd}/config/config2.js
export MM_PORT=8081
npm start
```

Currently, system is designed to have the "2" launch show up on screen 1, and the non-redirected go to screen 2 to work around bugs partially. If possible to fix bugs, I would then revert back to using 1 for screen 1, 2 for screen 2.

sdetweil, 2023-12-23:
can you try using all the variables from the sample config. I think some of them are required now.

::1  is localhost on ipv6
127.0.0.1 on ipv4

is your pi on an ipv6 network?

do
```
ip addr
```
from the command line

khassel, 2023-12-23:
I have no problems to set up 2 mm instances with a single-screen setup:

[image]

BKeyport, 2023-12-24:
Kassel: The problem is mostly a "technical" issue. Basically, you have to set up the 2nd monitor with the default config file (config.js) the way it's working for me. 

2nd monitor/instance uses config.js & mm.sh 
1st monitor/instance uses config2.js & mm2.sh 
The only difference in calling the mirror is the setting of the environment variable to call the correct config. 

It shouldn't matter which uses which, but the bugs above prevent it being done any other way.

BKeyport, 2023-12-24:
Sam: 

3: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether dc:a6:32:60:0e:4e brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.41/24 brd 192.168.0.255 scope global dynamic noprefixroute wlan0
       valid_lft 118019sec preferred_lft 118019sec
    inet6 fdeb:4b54:c1c1:244a:7b54:6243:c148:6c1c/64 scope global dynamic noprefixroute 
       valid_lft 1774sec preferred_lft 1774sec
    inet6 fe80::fd5a:3039:8824:6bd7/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever

Am I getting an IPv6 now? That would be new.

sdetweil, 2023-12-24:
you got an inet6 address ipv6

that comes from your dhcp server/router
it doesn't appear on the internet, just your local lan. the isp doesn't know

BKeyport, 2023-12-24:
Interesting. I'll have to look into it, I didn't think my router's DHCP was issuing IPV6.. But, really, that's a side quest to the problem at hand - if I set the address to 0.0.0.0, MagicMirror should be listening on 0.0.0.0:port no? 

```
root@officemirror:/home/bkey1970/PrivateBackup# ss -ltn
State       Recv-Q      Send-Q            Local Address:Port             Peer Address:Port      Process
LISTEN      0           511                     0.0.0.0:8080                  0.0.0.0:* (config.js launched Mirror instance, no environment vars set) 
LISTEN      0           511                       [::1]:8081                     [::]:* (config2.js launched Mirror instance, enviroment vars set) 
```

sdetweil, 2023-12-24:
0.0.0.0 means listen on all interfaces, without specifying what they are. so maybe 
it's got both v4 and v6 at the same time

BKeyport, 2023-12-24:
Look at the initial post - the ss command is indicating that the first mirror is listening on 0.0.0.0:8080, the 2nd is only on [::1]:8081. Nothing else has ports 8080/8081 open.

khassel, 2023-12-24:
> Kassel: The problem is mostly a "technical" issue. Basically, you have to set up the 2nd monitor with the default config file (config.js) the way it's working for me.
>
> 2nd monitor/instance uses config.js & mm.sh
> 1st monitor/instance uses config2.js & mm2.sh

I used a similar setup but without a second monitor and startet

`npm start` with `config.js` and `MM_CONFIG_FILE=config/config2.js npm start` and as you can see in my screenshot the `ss` command have both with `0.0.0.0:808x`

As already mentioned by Sam the ip6 address `[::1]:8081` seems to be the problem. For testing you could disable ip6 on your pi as e.g. described [here](https://gist.github.com/andreibosco/3badaac477446587bcd6751e186df446)

sdetweil, 2023-12-24:
they both cannot be using the same port

BKeyport, 2023-12-24:
OK - to make it clear - Both mirrors run - even if on the same port, so the 2nd mirror isn't opening up on top of the first. 

Once I disabled IPV6 using the method metioned above, I get this: 
```
0|MM1    | [24.12.2023 14:01.14.090] [WARN]  You're using a full whitelist configuration to allow for all IPs
0|MM1    | [24.12.2023 14:01.14.108] [ERROR] Whoops! There was an uncaught exception...
0|MM1    | [24.12.2023 14:01.14.113] [ERROR] Error: listen EADDRINUSE: address already in use 0.0.0.0:8080
0|MM1    |     at Server.setupListenHandle [as _listen2] (node:net:1740:16)
0|MM1    |     at listenInCluster (node:net:1788:12)
0|MM1    |     at doListen (node:net:1937:7)
0|MM1    |     at process.processTicksAndRejections (node:internal/process/task_queues:83:21) {
0|MM1    |   code: 'EADDRINUSE',
0|MM1    |   errno: -98,
0|MM1    |   syscall: 'listen',
0|MM1    |   address: '0.0.0.0',
0|MM1    |   port: 8080
```

Note my launching method is 
```
cd /home/bkey1970/MagicMirror
export MM_CONFIG_FILE=${pwd}/config/config2.js
export MM_PORT 8081
npm start
```

khassel, 2023-12-24:
> export MM_CONFIG_FILE=${pwd}/config/config2.js
> export MM_PORT 8081
> npm start

doing the same from mm folder after `npm start` for the first mirror start and this all works on my side ...

BKeyport, 2023-12-24:
define "works" -- is it not ignoring the above config items on your end, or are you getting dual mirrors and not testing further? Why I noticed it is because I want to have both mirrors available on my network for remote access as well as local display. Currently, I can't access the one using the env variable.

khassel, 2023-12-24:
> define "works" -- is it not ignoring the above config items on your end

yes the config vars are working on my side (see Picture in my post above), both mm starts on the pi screen and because of `ipWhitelist: [],` in both configs they are both reachable over `192.168.0.39:8080` and `192.168.0.39:8081` from outside

BKeyport, 2023-12-24:
OK - upon further testing, anytime I set the ```MM_CONFIG_FILE``` variable, it'll ignore the address and port, and default back to 127.0.0.1:8080 if IPV6 isn't available, if it is, it'll set [::1]:8080 - I can force it to another port with ```MM_PORT``` - on the same rules as above. This will happen for both my first and second instance. I'm lost as to why it's working for you. 

It will also ignore the ```electronOptions``` setting in the config. As I don't know where or how the environment variables are implemented, I have no idea how to look farther. I'm using the quarterly release - are you using the develop branch - I did mention earlier that ignoring the ```electronOptions``` was occuring, but we thought we had it fixed with a different EXPORT line.

khassel, 2023-12-24:
I'm using latest `develop` branch and `electronOptions` are working, I used them not to start fullscreen and to move the second mirror to the right

sdetweil, 2023-12-24:
@BKeyport what folder are you in when you do the exports?

the pwd part was just a shortcut to the current folder pathname.. (pwd means print working directory)  you said it needed braces  but should be parens when used in a bash script

if u echo the env var
echo $MM_CONFIG_FILE
you should see the whole correct path

BKeyport, 2023-12-25:
OK, Sam, I've gone to a direct link for everything - get bad config screen now
[image]
 
```
bkey1970@officemirror:~/MagicMirror $ export MM_CONFIG_FILE=/home/bkey1970/MagicMirror/config/config2.js
bkey1970@officemirror:~/MagicMirror $ echo $MM_CONFIG_FILE
/home/bkey1970/MagicMirror/config/config2.js
bkey1970@officemirror:~/MagicMirror $ npm run config:check

> magicmirror@2.26.0-develop config:check
> node js/check_config.js

[24.12.2023 16:23.55.191] [INFO]  Checking file...  /home/bkey1970/MagicMirror/config/config2.js
[24.12.2023 16:23.55.301] [INFO]  Your configuration file doesn't contain syntax errors :)
```

```
cd /home/bkey1970/MagicMirror
export MM_CONFIG_FILE=/home/bkey1970/MagicMirror/config/config2.js
# export MM_PORT=8081
export ELECTRON_DISABLE_GPU=1
DISPLAY=:0 npm start
```

```
/* Magic Mirror Config by Brendan Keyport. */
var config = {
	address: "0.0.0.0",
	port: 8081, 
	ipWhitelist: [], 
	language: "en",
	timeFormat: 12,
	units: "imperial",
	//logLevel: ["INFO", "LOG", "WARN", "ERROR", "DEBUG"],
	modules: [
{
  module: "MMM-EmbedURL",
  position: "top_left",
  config: {
    updateInterval: 600000,
    animationSpeed: 0, 
    attributes: [
        "frameborder=0",
    ],
    embed: [
        "http://192.168.0.7/picture/1/frame/"
    ]
  },
},
{
  module: "MMM-EmbedURL",
  position: "top_right",
  config: {
    updateInterval: 0,
    animationSpeed: 0, 
    attributes: [
        "frameborder=0",
    ],
    embed: [
        "http://192.168.0.7/picture/2/frame/"
    ]
  },
},
{
  module: "MMM-EmbedURL",
  position: "bottom_left",
  config: {
    updateInterval: 0,
    animationSpeed: 0, 
    attributes: [
[... 105 lines cut ...]
1|MM2    | [55037:1224/162613.141852:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
1|MM2    | [55037:1224/162613.143469:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
1|MM2    | [55037:1224/162613.146873:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
1|MM2    | [55037:1224/162613.148676:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
1|MM2    | [55037:1224/162613.150590:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
1|MM2    | [55037:1224/162613.152367:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
1|MM2    | [55037:1224/162613.154794:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
1|MM2    | [55037:1224/162613.155408:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
1|MM2    | [55037:1224/162613.156815:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
0|MM1    | [24.12.2023 16:26.13.353] [LOG]   Launching application.
0|MM1    | [55094:1224/162614.543038:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
0|MM1    | [55094:1224/162614.543667:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
0|MM1    | [55094:1224/162614.556042:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
0|MM1    | [55094:1224/162614.556428:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
0|MM1    | [55094:1224/162614.558879:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
0|MM1    | [55094:1224/162614.559331:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
0|MM1    | [55094:1224/162614.561950:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
0|MM1    | [55094:1224/162614.568131:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
0|MM1    | [55094:1224/162614.568579:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
0|MM1    | [55094:1224/162614.568959:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
0|MM1    | [55094:1224/162614.571308:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
0|MM1    | [55094:1224/162614.571716:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
0|MM1    | [55094:1224/162614.572088:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
0|MM1    | [55094:1224/162614.573264:ERROR:gbm_wrapper.cc(253)] Failed to export buffer to dma_buf: No such file or directory (2)
```

BKeyport, 2023-12-25:
```
bkey1970@officemirror:~ $ sudo ss -ltn
State                  Recv-Q                 Send-Q                                  Local Address:Port                                   Peer Address:Port                 Process
LISTEN                 0                      511                                           0.0.0.0:8081                                        0.0.0.0:*
LISTEN                 0                      511                                           0.0.0.0:8080                                        
```

sdetweil, 2023-12-25:
well, the code in electron.js  doesn't use the MM_PORT env variable, only the one from the config file. 

```
 mainWindow.loadURL(`${prefix}${address}:${config.port}`);
```

the env variable is saved in app.js, but never referenced..

BKeyport, 2023-12-25:
to me, there's clearly a processing issue with the MM_CONFIG_FILE set. It's ignoring parts of the file set within each and every time. 

as for MM_PORT - you'll notice I have it commented out above - it's reading it from the config file. That was from an earlier test trying to get it accessible from other machines.

BKeyport, 2023-12-25:
BTW, Merry Bah Humbug to you both.

sdetweil, 2023-12-25:
lol.nah, its just stuff.  had a great time w young grandkids this morning.  my 3d printing of pi5 case is going along.

rejas, 2023-12-26:
totally unrelated question: what 3d printer modell do you have?

sdetweil, 2023-12-26:
I have an Ender 3 V2. I picked it up used earlier this year for another project. I use Octoprint on a pi02w to manage the print

khassel, 2023-12-26:
> as for MM_PORT - you'll notice I have it commented out above - it's reading it from the config file.

can you test [my PR](https://github.com/MichMich/MagicMirror/pull/3304/files) ? For Sam it did not work but in my tests it did ... thanks!

BKeyport, 2023-12-26:
MM_PORT actually did work for me prior. I don't understand what I'm supposed to test here. I'd prefer to not use it at all, and actually have the config2.js read correctly. 😉

sdetweil, 2023-12-26:
get the develop branch
see https://forum.magicmirror.builders/topic/14327/testing-new-fixes-or-solving-current-problems-with-next-release-code

and test the MM_CONFIG_FILE env var

BKeyport, 2023-12-26:
I know _how_ to do it, I just don't know _why_, sam. That particular bug didn't exist for me.

BKeyport, 2023-12-26:
or, are you saying MM_CONFIG_FILE has been updated too?

BKeyport, 2023-12-27:
So, here's where I'm at: Using develop branch for all tests for consistency. 

```
cd /home/bkey1970/MagicMirror
# works - comes up on [::1]:8080 - ignores address, port in config2.js 
export MM_CONFIG_FILE=${pwd}/config/config2.js

# Fails to error screen - but 0.0.0.0:8081 - ignores modules in config2.js 
# export MM_CONFIG_FILE=$(pwd)/config/config2.js
# export MM_CONFIG_FILE=/home/bkey1970/MagicMirror/config/config2.js

# works with working config above. moves device to [::1]:8081
# export MM_PORT=8081

export ELECTRON_DISABLE_GPU=1
DISPLAY=:0 npm start
```

Next move?

sdetweil, 2023-12-27:
after you do 
export MM_CONFIG_FILE=${pwd}/config/config2.js

can you do
echo $MM_CONFIG_FILE

are you doing this in batch file.sh) or from command line?

BKeyport, 2023-12-27:
I'm doing both as needed to see results. Command line test:
```
bkey1970@officemirror:~/MagicMirror $ export MM_CONFIG_FILE=${pwd}/config/config2.js
bkey1970@officemirror:~/MagicMirror $ echo $MM_CONFIG_FILE
/config/config2.js
```

sdetweil, 2023-12-27:
${pwd} is wrong ( braces)

do $(pwd)    (parens)

BKeyport, 2023-12-27:
see my notes - (pwd) fails to error screen, but loads the address/port from the config. 

```
bkey1970@officemirror:~/PrivateBackup $ export MM_CONFIG_FILE=$(pwd)/config/config2.js
bkey1970@officemirror:~/PrivateBackup $ echo $MM_CONFIG_FILE
/home/bkey1970/PrivateBackup/config/config2.js
```

We're running in circles - the issue seems to come down to config file is being partially read *unless* you strip off the leading parts of the path ```config/config2.js``` is working completely, nothing is being ignored from what I'm seeing.

sdetweil, 2023-12-27:
cool.. so it's this line of code in app.js
```
const configFilename = path.resolve(global.configuration_file || `${global.root_path}/config/config.js`);
```

sdetweil, 2023-12-27:
so,maybe the MM_CONFIG_FILE must be someplace in the MagicMirror folder tree
and you only specify the path from there. 

I debugged this tho a couple weeks ago and it returned the full path when it was specified, and read the correct file.

khassel, 2023-12-27:
> get the develop branch

the patch is not merged yet so not on `develop` ...

khassel, 2023-12-27:
>  `config/config2.js` is working completely

o.k., this explains why my tests were working because I always used `MM_CONFIG_FILE=config/config2.js` (without pwd)

> `const configFilename = path.resolve(global.configuration_file || `${global.root_path}/config/config.js`);`

I thought it was a requirement that `config.js` must be located underneath the mm folder

BKeyport, 2023-12-27:
Even if it's a requirement, /home/<user>/MagicMirror/config/config.js should equal config/config.js if started in the MagicMirror directory, no?

khassel, 2023-12-27:
yes, will look into this later if there is a simple fix

BKeyport, 2023-12-27:
I know we're coming up on the next quarterly, so low priority. Was able to get things working properly with the shortened variable - so, that's a documented work around until such time as it's fixed..

BKeyport, 2023-12-29:
noticed a doc suggestion in #3304 so I've updated my suggested docs to cover said suggestion.

khassel, 2024-01-03:
@BKeyport the documentation PR is on the way, can we close this here or is there still something to discuss?

closed 2024-01-03 by BKeyport