MagicMirrorOrg/MagicMirror #3307: [ERROR] TypeError: Cannot read properties of undefined (reading 'on')

Prototype1997, 2023-12-26:
Hey Guys,

i have taken a clean Install on a Raspberry 2b and getting this follwing Error in PM2
`

0|MagicMirror  | [26.12.2023 22:01.53.214] [LOG]   Starting MagicMirror: v2.25.0
0|MagicMirror  | [26.12.2023 22:01.53.283] [ERROR] TypeError: Cannot read properties of undefined (reading 'on')
0|MagicMirror  |     at Object.<anonymous> (/home/alex/MagicMirror/js/electron.js:139:5)
0|MagicMirror  |     at Module._compile (node:internal/modules/cjs/loader:1241:14)
0|MagicMirror  |     at Module._extensions..js (node:internal/modules/cjs/loader:1295:10)
0|MagicMirror  |     at Module.load (node:internal/modules/cjs/loader:1091:32)
0|MagicMirror  |     at Module._load (node:internal/modules/cjs/loader:938:12)
0|MagicMirror  |     at Object.<anonymous> (/usr/local/lib/node_modules/pm2/lib/ProcessContainerFork.js:33:23)
0|MagicMirror  |     at Module._compile (node:internal/modules/cjs/loader:1241:14)
0|MagicMirror  |     at Module._extensions..js (node:internal/modules/cjs/loader:1295:10)
0|MagicMirror  |     at Module.load (node:internal/modules/cjs/loader:1091:32)
0|MagicMirror  |     at Module._load (node:internal/modules/cjs/loader:938:12)
0|MagicMirror  |     at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:83:12)
0|MagicMirror  |     at node:internal/main/run_main_module:23:47
0|MagicMirror  | [26.12.2023 22:01.53.317] [ERROR] Whoops! There was an uncaught exception...
0|MagicMirror  | [26.12.2023 22:01.53.321] [ERROR] TypeError: Cannot read properties of undefined (reading 'on')
0|MagicMirror  |     at Object.<anonymous> (/home/alex/MagicMirror/js/electron.js:139:5)
0|MagicMirror  |     at Module._compile (node:internal/modules/cjs/loader:1241:14)
0|MagicMirror  |     at Module._extensions..js (node:internal/modules/cjs/loader:1295:10)
0|MagicMirror  |     at Module.load (node:internal/modules/cjs/loader:1091:32)
0|MagicMirror  |     at Module._load (node:internal/modules/cjs/loader:938:12)
0|MagicMirror  |     at Object.<anonymous> (/usr/local/lib/node_modules/pm2/lib/ProcessContainerFork.js:33:23)
0|MagicMirror  |     at Module._compile (node:internal/modules/cjs/loader:1241:14)
0|MagicMirror  |     at Module._extensions..js (node:internal/modules/cjs/loader:1295:10)
0|MagicMirror  |     at Module.load (node:internal/modules/cjs/loader:1091:32)
0|MagicMirror  |     at Module._load (node:internal/modules/cjs/loader:938:12)
0|MagicMirror  |     at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:83:12)
0|MagicMirror  |     at node:internal/main/run_main_module:23:47
0|MagicMirror  | [26.12.2023 22:01.53.327] [ERROR] MagicMirror² will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
0|MagicMirror  | [26.12.2023 22:01.53.331] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
PM2            | App [MagicMirror:0] exited with code [0] via signal [SIGINT]
PM2            | App [MagicMirror:0] starting in -fork mode-

`
does anybody have a solution for this Problem?

Thanks for your Help!

khassel, 2023-12-26:
please provide output of following commands:
- `uname -a`
- `node -v`
- `cat /etc/os-release`

Prototype1997, 2023-12-27:
> please provide output of following commands:
> 
> * `uname -a`
> * `node -v`
> * `cat /etc/os-release`

 uname -a
Linux raspberrypi 6.1.65-v7+ #1703 SMP Tue Dec  5 16:20:29 GMT 2023 armv7l GNU/Linux

node -v
v20.8.0

cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 11 (bullseye)"
NAME="Raspbian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Prototype1997, 2023-12-27:
Maybe as an Information,
i can start MagicMirror over NPM, in this case i have no Errors. 
But when i try to start MagicMirror with PM2 then i get the Errors.

Prototype1997, 2023-12-27:
After an update from PM2 i got everthing Working right Now. Thanks for your help !

closed 2023-12-27 by Prototype1997

sdetweil, 2023-12-27:
how did you install the first time?  manual or use my script?