MagicMirrorOrg/MagicMirror #3310: MagicMirror stopped working after update. Help!

websterjazz, 2023-12-28:
Last login: Fri Dec 29 02:14:42 2023 from 192.168.29.127
pi@Smart-Mirror:~ $ cd MagicMirror && npm start run

> magicmirror@2.19.0 start /home/pi/MagicMirror
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js "run"

[29.12.2023 02:17.13.809] [LOG]   Starting MagicMirror: v2.19.0
[29.12.2023 02:17.13.818] [LOG]   Loading config ...
[29.12.2023 02:17.13.826] [LOG]   Loading module helpers ...
[29.12.2023 02:17.13.829] [LOG]   No helper found for module: alert.
[29.12.2023 02:17.13.847] [LOG]   Initializing new module helper ...
[29.12.2023 02:17.13.848] [LOG]   Module helper loaded: updatenotification
[29.12.2023 02:17.13.850] [LOG]   No helper found for module: clock.
[29.12.2023 02:17.14.029] [LOG]   Initializing new module helper ...
[29.12.2023 02:17.14.030] [LOG]   Module helper loaded: calendar
[29.12.2023 02:17.14.032] [LOG]   No helper found for module: weather.
[29.12.2023 02:17.14.034] [LOG]   No helper found for module: MMM-EyeCandy.
[29.12.2023 02:17.14.359] [LOG]   Initializing new module helper ...
[29.12.2023 02:17.14.360] [LOG]   Module helper loaded: MMM-EARTH
[29.12.2023 02:17.14.361] [LOG]   No helper found for module: MMM-Cursor.
[29.12.2023 02:17.14.363] [LOG]   No helper found for module: MMM-EasyBack.
[29.12.2023 02:17.14.365] [LOG]   Initializing new module helper ...
[29.12.2023 02:17.14.367] [LOG]   Module helper loaded: MMM-Volume
[29.12.2023 02:17.14.371] [ERROR] WARNING! Could not load config file. Starting with default configuration. Error found: Error: Cannot find module 'request'
Require stack:
- /home/pi/MagicMirror/modules/MMM-ImagesPhotos/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
-
[29.12.2023 02:17.14.372] [LOG]   Loading module helpers ...
[29.12.2023 02:17.14.373] [LOG]   No helper found for module: alert.
[29.12.2023 02:17.14.374] [LOG]   Initializing new module helper ...
[29.12.2023 02:17.14.375] [LOG]   Module helper loaded: updatenotification
[29.12.2023 02:17.14.376] [LOG]   No helper found for module: clock.
[29.12.2023 02:17.14.377] [LOG]   Initializing new module helper ...
[29.12.2023 02:17.14.378] [LOG]   Module helper loaded: calendar
[29.12.2023 02:17.14.379] [LOG]   No helper found for module: weather.
[29.12.2023 02:17.14.380] [LOG]   No helper found for module: MMM-EyeCandy.
[29.12.2023 02:17.14.381] [LOG]   Initializing new module helper ...
[29.12.2023 02:17.14.382] [LOG]   Module helper loaded: MMM-EARTH
[29.12.2023 02:17.14.383] [LOG]   No helper found for module: MMM-Cursor.
[29.12.2023 02:17.14.384] [LOG]   No helper found for module: MMM-EasyBack.
[29.12.2023 02:17.14.385] [LOG]   Initializing new module helper ...
[29.12.2023 02:17.14.386] [LOG]   Module helper loaded: MMM-Volume
[29.12.2023 02:17.14.388] [ERROR] App threw an error during load
[29.12.2023 02:17.14.390] [ERROR] Error: Cannot find module 'request'
Require stack:
- /home/pi/MagicMirror/modules/MMM-ImagesPhotos/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
-
    at Module._resolveFilename (node:internal/modules/cjs/loader:940:15)
    at Function.n._resolveFilename (node:electron/js2c/browser_init:249:1105)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Module._load (node:internal/modules/cjs/loader:785:27)
    at Function.c._load (node:electron/js2c/asar_bundle:5:13331)
    at Module.require (node:internal/modules/cjs/loader:1012:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-ImagesPhotos/node_helper.js:10:15)
    at Module._compile (node:internal/modules/cjs/loader:1116:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1169:10)
[29.12.2023 02:17.14.392] [ERROR] Whoops! There was an uncaught exception...
[29.12.2023 02:17.14.398] [ERROR] Error: Cannot find module 'request'
Require stack:
- /home/pi/MagicMirror/modules/MMM-ImagesPhotos/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
-
    at Module._resolveFilename (node:internal/modules/cjs/loader:940:15)
    at Function.n._resolveFilename (node:electron/js2c/browser_init:249:1105)
    at Function.Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at Module._load (node:internal/modules/cjs/loader:785:27)
    at Function.c._load (node:electron/js2c/asar_bundle:5:13331)
    at Module.require (node:internal/modules/cjs/loader:1012:19)
    at require (node:internal/modules/cjs/helpers:102:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/MMM-ImagesPhotos/node_helper.js:10:15)
    at Module._compile (node:internal/modules/cjs/loader:1116:14)
    at Object.Module._extensions..js (node:internal/modules/cjs/loader:1169:10) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [
    '/home/pi/MagicMirror/modules/MMM-ImagesPhotos/node_helper.js',
    '/home/pi/MagicMirror/js/app.js',
    '/home/pi/MagicMirror/js/electron.js',
    '/home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js',
    undefined
  ]
}
[29.12.2023 02:17.14.402] [ERROR] MagicMirror² will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
[29.12.2023 02:17.14.402] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
[29.12.2023 02:17.14.712] [LOG]   Launching application.

sdetweil, 2023-12-28:
first this should come to the forums.as it's ot a MagicMirror problem.

the module does not document the use of the request library which we removed in ,version 2.16 , now 2 years ago .  the short term fix is to go to the module folder and do
```
npm install request
```

khassel, 2023-12-28:
> first this should come to the forums. as it's not a MagicMirror problem.

yes, and old mm version and wrong start command, so closing here.

closed 2023-12-28 by khassel