MagicMirrorOrg/MagicMirror #3312: Error in Calender - Size issue

TDMPro2K, 2023-12-31:
Hi All, after some advice. I have a shared calendar which I link to my mirror.
It appears to now error out and just says Error in Calendar module. Check logs

I can link a new one fine and if I delete the old events from the calender it works. So it seems to be a issue with the size of the Ical.
It 500kb ish.
So is there a way to stop it from looking prior to xx date?
I cant delete the old events as others use them for reference. 

Also I cant find the error logs for the calendar. Any advice welcome

sdetweil, 2023-12-31:
the errors are printed to the place where you start mm, npm start

if you use pm2 to launch mm  then the command

pm2 logs --lines=50

might show them

more than likely this is not a size issue but a networking issue.u

TDMPro2K, 2023-12-31:
It not a network issue, if I delete all events and leave the last 6month the it’s fine and works.

sdetweil, 2023-12-31:
can you show me the error from the startup messages?

we don't control what the calendar sends as events. but the runtime might have run out of memory.

sdetweil, 2023-12-31:
# Use 4096 MB as max memory for Node
export NODE_OPTIONS=--max_old_space_size=4096

before you do the npm  start

if using pm2 to launch, I would test first
```
pm2 stop all
cd ~/MagicMirror
npm start
```
messages will be here, ctrl-q to stop mm

TDMPro2K, 2023-12-31:
Where do i put 
export NODE_OPTIONS=--max_old_space_size=4096

Just paste in terminal? Or do I need to put it in a file

TDMPro2K, 2023-12-31:
I will paste the error logs when I get back home

sdetweil, 2023-12-31:
try pasto g and starting mm manually w npm start... to test

if it works and u are using pm2, from my install script then 
```bash
~/MagicMirror/installers/mm.sh
```

before the npm start line

TDMPro2K, 2023-12-31:
So i tried the export 
NODE_OPTIONS=--max_old_space_size=4096
and mm wouldn't start. it said allocation failure.

the mm-error-log
[1762:1231/230723.028797:ERROR:viz_main_impl.cc(186)] Exiting GPU process due to errors during initialization
[1799:1231/230723.385909:ERROR:gpu_memory_buffer_support_x11.cc(44)] dri3 extension not supported.
[1767:1231/230723.466337:ERROR:command_buffer_proxy_impl.cc(128)] ContextResult::kTransientFailure: Failed to send GpuControl.CreateCommandBuffer.
[31.12.2023 23:07.36.115] [ERROR] Calendar Error. Could not fetch calendar:  https://tomycalendar.com/calendar.ics Error: ERROR when trying to convert to ISOString
    at Object.END (/home/pi/MagicMirror/node_modules/node-ical/ical.js:597:21)
    at Object.handleObject (/home/pi/MagicMirror/node_modules/node-ical/ical.js:644:39)
    at Object.parseLines (/home/pi/MagicMirror/node_modules/node-ical/ical.js:696:18)
    at Object.parseICS (/home/pi/MagicMirror/node_modules/node-ical/ical.js:732:18)
    at sync.parseICS (/home/pi/MagicMirror/node_modules/node-ical/node-ical.js:198:15)
    at autodetect.parseICS (/home/pi/MagicMirror/node_modules/node-ical/node-ical.js:229:17)
    at /home/pi/MagicMirror/modules/default/calendar/calendarfetcher.js:73:18
    at process.processTicksAndRejections (node:internal/process/task_queues:96:5)

TDMPro2K, 2023-12-31:
it i make the calendar smaller it works... So it has to be the size, and it timing out other reading the entries. Thats the only thing I can put it down to.

sdetweil, 2023-12-31:
fix for isostring coming in release tomorrow.

sdetweil, 2023-12-31:
it's a bug in the rrule.between when the cal event is expired. returns garbage dates

TDMPro2K, 2024-01-01:
Ok, I will wait for the update. 

Cheers

sdetweil, 2024-01-01:
@TDMPro2K   new release is available..
use my upgrade script

TDMPro2K, 2024-01-06:
HI just thought i would report back. So New years day and up to today, it worked without me doing anything.
Today it stopped working again and gave the error again. So I have ran the update to the latest version and still get the same error.

06.01.2024 09:47.54.271] [ERROR] Calendar Error. Could not fetch calendar:  https://linktomycalendar/calendar.ics Error: ERROR when trying to convert to ISOString
    at Object.END (/home/pi/MagicMirror/node_modules/node-ical/ical.js:597:21)
    at Object.handleObject (/home/pi/MagicMirror/node_modules/node-ical/ical.js:644:39)
    at Object.parseLines (/home/pi/MagicMirror/node_modules/node-ical/ical.js:696:18)
    at Object.parseICS (/home/pi/MagicMirror/node_modules/node-ical/ical.js:732:18)
    at sync.parseICS (/home/pi/MagicMirror/node_modules/node-ical/node-ical.js:198:15)
    at autodetect.parseICS (/home/pi/MagicMirror/node_modules/node-ical/node-ical.js:229:17)
    at /home/pi/MagicMirror/modules/default/calendar/calendarfetcher.js:73:18
    at process.processTicksAndRejections (node:internal/process/task_queues:96:5)

sdetweil, 2024-01-06:
@TDMPro2K thank you. this is a failure in the parser, thanks for the stack trace.

would u be willing to share your cal url
if so, please use a chat on forums or discord (I'm Sam there)
or download the ICS file
curl -sL url >somefile.txt
and mail it to me, same userid at gmail.

TDMPro2K, 2024-01-08:
@sdetweil I have Emailed sir.

Cheers

sdetweil, 2024-01-08:
thanks.. this is another case of MS has messed everybody up again..

they have started using custom timezones, which  none of our tools support. 

I built in a lookup table 4 years ago to match the MS TZ with the official  iana TZ names.. (which all of the code libraries use) 

but now we have trash again

from the cal 
```
"(UTC+00:00) Dublin, Edinburgh, Lisbon, London"
```
in the node-ical parser i get the windows timezones from here 
https://github.com/unicode-org/cldr/blob/main/common/supplemental/windowsZones.xml

i've tried some workarounds, but it continues on..

TDMPro2K, 2024-01-08:
Ok, I will look at doing a sync and switch to Google calender and sync the two some how.Sent from my iPhone

https://github.com/unicode-org/cldr/blob/main/common/supplemental/windowsZones.xml

closed 2024-10-13 by github-actions[bot]