MagicMirrorOrg/MagicMirror #3318: MMM shows only Undefined

oguzhanuslu, 2024-01-01:
I installed MMM and able to run it. The problem is MMM shows only Undefined and it does not matter have 3rd party modules or not. I tried solutions on the internet but did not work. What should I do?

To addition the information above, I also changed the `socketclient.js`. You can find the codes below.

MMM version: 2.25.0
Node JS version: 20.10.0
Operating System: Windows 10

[image]

```
var MMSocket = function(moduleName) {
	var self = this;

	if (typeof moduleName !== "string") {
		throw new Error("Please set the module name for the MMSocket.");
	}

	self.moduleName = moduleName;

	// Private Methods
	self.socket = io("/" + self.moduleName);
	var notificationCallback = function() {};

	var onevent = self.socket.onevent;
	self.socket.onevent = function(packet) {
		var args = packet.data || [];
		onevent.call(this, packet);    // original call
		packet.data = ["*"].concat(args);
		onevent.call(this, packet);      // additional call to catch-all
	};

	// register catch all.
	self.socket.on("*", function(notification, payload) {
		if (notification !== "*") {
			notificationCallback(notification, payload);
		}
	});

	// Public Methods
	this.setNotificationCallback = function(callback) {
		notificationCallback = callback;
	};

	this.sendNotification = function(notification, payload) {
		if (typeof payload === "undefined") {
			payload = {};
		}
		self.socket.emit(notification, payload);
	};
};
```

sdetweil, 2024-01-01:
did you do the two additional npm install steps? in vendor and fonts, per the documentation.

sdetweil, 2024-01-01:
why did u change socketClient?

oguzhanuslu, 2024-01-01:
> why did u change socketClient?

1. As I remember I didn't do that but how do I check?
2. I don't know exactly how it works, I saw the solution (don't remember which website) and some positive comments and I did it.

sdetweil, 2024-01-01:
in the MagicMirror folder do
```bash
git status
```
to see the condition of all the files provided against the source repo versions

but again  what lead you to do this? for what proposed fix?

oguzhanuslu, 2024-01-01:
> in the MagicMirror folder do
> 
> ```shell
> git status
> ```
> 
> to see the condition of all the files provided against the source repo versions
> 
> but again what lead you to do this? for what proposed fix?

I am not a good computer person, sorry about that but I saw and tried it. I have no clue what I did.

$ git status
On branch master
Your branch is up to date with 'origin/master'.

```
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        modified:   css/main.css
        modified:   js/socketclient.js
        modified:   package.json

no changes added to commit (use "git add" and/or "git commit -a")
```

sdetweil, 2024-01-01:
ok, 1st ALL css changes go in css/custom.css
(if the file doesn't exist, cxreate it) 

please put back socketclient
```
git checkout js/socketclient.js
```

also do
```
cd  MagicMirror\fonts 
npm install
cd ..\vendor
npm install
```

then restart  MagicMirror

oguzhanuslu, 2024-01-01:
It works!! Thank you so much.

closed 2024-01-01 by oguzhanuslu

sdetweil, 2024-01-01:
closing,  please make sure to follow all the instructions 
https://docs.magicmirror.builders/getting-started/installation.html#other-operating-systems