MagicMirrorOrg/MagicMirror #3320: Add CHANGELOG entry for dependabot PRs

rejas, 2024-01-01:
Currently our checks complain about the dependabot PRs since they dont add a CHANGELOG entry.

Maybe someone can look into https://github.com/dangoslen/dependabot-changelog-helper
and see if it can be used to turn the PR-checks green.

khassel, 2024-01-01:
can take a look but do we really need this?

I'm fine with the one failing test and we can merge these PR's if all other tests are green.

I don't need every version update in the changelog, this will only blew up this file with unneeded things. For me it's o.k. to mention only significant updates as e.g. `electron` and have an entry `dependencies updated` for the normal updates (what we are doing already).

And I think we don't need this for the whole dev dependencies.

khassel, 2024-01-01:
as an alternative approach we could add the label `Skip Changelog` to the `dependabot` PR's to avoid the failed tests.

rejas, 2024-01-01:
> as an alternative approach we could add the label `Skip Changelog` to the `dependabot` PR's to avoid the failed tests.

I am totally fine with this approach

khassel, 2024-01-08:
still not working, the `Skip Changelog` label is missing so check fails.

rejas, 2024-01-08:
maybe the PR needs to find its way to master first?

rejas, 2024-01-14:
I you want to, you can cherrypick it. I myself wouldnt mind  aiting for the next release branch.

closed 2024-04-01 by rejas