MagicMirrorOrg/MagicMirror #3323: MagicMirror v2.26.0 won't start with pm2 but running fine, if starting manually - undefined reading disableHardwareAcceleration

eckonator, 2024-01-01:
Hello,
I have just updated my MagicMirror to version 2.26.0. Afterwards I started it manually in the MagicMirror directory with

`npm start`

This worked perfectly.

Now I have also updated **pm2** to the latest version: `v5.3.0`
Unfortunately, my MagicMirror does not manage to start with pm2.

The error log gives me the following error message:

```
0|MagicMir | [01.01.2024 19:32.53.347] [ERROR] TypeError: Cannot read properties of undefined (reading 'disableHardwareAcceleration')
0|MagicMir |     at Object.<anonymous> (/home/pi/MagicMirror/js/electron.js:15:6)
0|MagicMir |     at Module._compile (node:internal/modules/cjs/loader:1356:14)
0|MagicMir |     at Module._extensions..js (node:internal/modules/cjs/loader:1414:10)
0|MagicMir |     at Module.load (node:internal/modules/cjs/loader:1197:32)
0|MagicMir |     at Module._load (node:internal/modules/cjs/loader:1013:12)
0|MagicMir |     at Object.<anonymous> (/usr/lib/node_modules/pm2/lib/ProcessContainerFork.js:33:23)
0|MagicMir |     at Module._compile (node:internal/modules/cjs/loader:1356:14)
0|MagicMir |     at Module._extensions..js (node:internal/modules/cjs/loader:1414:10)
0|MagicMir |     at Module.load (node:internal/modules/cjs/loader:1197:32)
0|MagicMir |     at Module._load (node:internal/modules/cjs/loader:1013:12)
0|MagicMir |     at Function.executeUserEntryPoint [as runMain] (node:internal/modules/run_main:128:12)
0|MagicMir |     at node:internal/main/run_main_module:28:49
0|MagicMir | [01.01.2024 19:32.53.353] [ERROR] MagicMirror² will not quit, but it might be a good idea to check why this happened. Maybe no internet connection?
0|MagicMir | [01.01.2024 19:32.53.356] [ERROR] If you think this really is an issue, please open an issue on GitHub: https://github.com/MichMich/MagicMirror/issues
```

What can I do to make my MagicMirror start with pm2 again?

sdetweil, 2024-01-01:
I sent an email back

sdetweil, 2024-01-01:
do this
```
bash -c "$(curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/upgradepm2.sh)"
```

eckonator, 2024-01-01:
Thanks, this worked for me!

closed 2024-01-01 by eckonator

sdetweil, 2024-01-01:
that makes  3 users now. thanks for the feedback