MagicMirrorOrg/MagicMirror #3325: PB mise à jour

jeff44850, 2024-01-01:
Bonjour
Bonne et heureuse année pour commencer 😉
Suite à la mise à jour de MM 
j'ai ce message et mm ne démarre donc pas 
À quoi est-ce lié ? Mise à jour électron ??
Bon courage 

pi@raspberrypi:~ $ cd ~/MagicMirror
pi@raspberrypi:~/MagicMirror $ npm start

> magicmirror@2.26.0 start
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

[01.01.2024 21:38.52.139] [LOG]   Starting MagicMirror: v2.26.0
[01.01.2024 21:38.52.159] [LOG]   Loading config ...
[01.01.2024 21:38.52.168] [DEBUG] config template file not exists, no envsubst
[01.01.2024 21:38.52.192] [LOG]   Loading module helpers ...
[01.01.2024 21:38.52.197] [LOG]   No helper found for module: clock.
[01.01.2024 21:38.52.812] [LOG]   Initializing new module helper ...
[01.01.2024 21:38.52.814] [LOG]   Module helper loaded: calendar
[01.01.2024 21:38.52.966] [LOG]   Initializing new module helper ...
[01.01.2024 21:38.52.969] [LOG]   Module helper loaded: MMM-Weather
[01.01.2024 21:38.52.972] [LOG]   No helper found for module: MMM-TomTomTraffic.
[01.01.2024 21:38.54.003] [LOG]   Initializing new module helper ...
[01.01.2024 21:38.54.006] [LOG]   Module helper loaded: EXT-Librespot
[01.01.2024 21:38.54.018] [LOG]   Initializing new module helper ...
[01.01.2024 21:38.54.019] [LOG]   Module helper loaded: MMM-GoogleAssistant
[01.01.2024 21:38.54.147] [LOG]   Initializing new module helper ...
[01.01.2024 21:38.54.148] [LOG]   Module helper loaded: MMM-News
[01.01.2024 21:38.54.150] [LOG]   No helper found for module: MMM-TouchNotifications.
[01.01.2024 21:38.54.165] [LOG]   Initializing new module helper ...
[01.01.2024 21:38.54.166] [LOG]   Module helper loaded: Gateway
[01.01.2024 21:38.54.193] [ERROR] (node:4672) UnhandledPromiseRejectionWarning: Error: Cannot find module 'command-exists'
Require stack:
- /home/pi/MagicMirror/modules/default/updatenotification/update_helper.js
- /home/pi/MagicMirror/modules/default/updatenotification/node_helper.js
- /home/pi/MagicMirror/js/app.js
- /home/pi/MagicMirror/js/electron.js
- /home/pi/MagicMirror/node_modules/electron/dist/resources/default_app.asar/main.js
- 
    at node:internal/modules/cjs/loader:1084:15
    at Function.<anonymous> (node:electron/js2c/browser_init:2:117419)
    at Module._resolveFilename (/home/pi/MagicMirror/node_modules/module-alias/index.js:49:29)
    at node:internal/modules/cjs/loader:929:27
    at Function._load (node:electron/js2c/asar_bundle:2:13327)
    at Module.require (node:internal/modules/cjs/loader:1150:19)
    at require (node:internal/modules/cjs/helpers:110:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/updatenotification/update_helper.js:3:23)
    at Module._compile (node:internal/modules/cjs/loader:1271:14)
    at Object..js (node:internal/modules/cjs/loader:1326:10)
    at Module.load (node:internal/modules/cjs/loader:1126:32)
    at node:internal/modules/cjs/loader:967:12
    at Function._load (node:electron/js2c/asar_bundle:2:13327)
    at Module.require (node:internal/modules/cjs/loader:1150:19)
    at require (node:internal/modules/cjs/helpers:110:18)
    at Object.<anonymous> (/home/pi/MagicMirror/modules/default/updatenotification/node_helper.js:4:22)
(Use `electron --trace-warnings ...` to show where the warning was created)
[01.01.2024 21:38.54.195] [ERROR] (node:4672) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
[4672:0101/213855.146418:ERROR:object_proxy.cc(590)] Failed to call method: org.freedesktop.portal.Settings.Read: object_path= /org/freedesktop/portal/desktop: org.freedesktop.portal.Error.NotFound: Requested setting not found

KristjanESPERANTO, 2024-01-01:
How did you perform the update? By running `git pull && npm run install-mm`?

jeff44850, 2024-01-01:
Oui tout à fait

KristjanESPERANTO, 2024-01-01:
Which node version do you have? Run `node -v` to check.

jeff44850, 2024-01-01:
Je suis sur celle-ci
@raspberrypi:~ $ node -v
v18.17.1

KristjanESPERANTO, 2024-01-01:
Please try a minimal configuration (e.g. only with the clock). Does the mirror work with it?

I am afraid that it is caused by a 3rd party module.

khassel, 2024-01-01:
@jeff44850 we expect english here and I think everyone is able to use google translate

Looking at the error message you have to run `npm run install-mm` again, see also https://forum.magicmirror.builders/topic/18038/merge-conflict-detected-in-your-package-json

jeff44850, 2024-01-02:
Good evening problem solved for me with cd ~/MagicMirror git reset --hard git pull npm install Have a good evening

closed 2024-01-02 by khassel

sdetweil, 2024-01-02:
so the git pull is not succeeding..