MagicMirrorOrg/MagicMirror #3328: Add additional infos into log

KristjanESPERANTO, 2024-01-01:
When troubleshooting in the forum or in the GitHub issues, we often get screenshots or photos [like this](https://private-user-images.githubusercontent.com/62189127/293600498-125060ef-8025-4074-89b2-a377c2ceedaa.JPG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDQxNDQxMTEsIm5iZiI6MTcwNDE0MzgxMSwicGF0aCI6Ii82MjE4OTEyNy8yOTM2MDA0OTgtMTI1MDYwZWYtODAyNS00MDc0LTg5YjItYTM3N2MyY2VlZGFhLkpQRz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMDElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTAxVDIxMTY1MVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWNjYTEyM2I4MzY1MmNlODE3MzEwNDc3M2MzZTVkZTQ1MmM3MDcyMzFkZDBjZjI2YTg3ZmFlN2VhNjI5OThkNGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.600iVgD3CIfvDO9crbsst4-wuykNxTK9qz9qQ1oPDLY).

Wouldn't it be helpful if we could add some more information like node version or OS into the log. Then we wouldn't have to ask the users for it.

Something like this:

```sh
[01.01.2024 22:22.15.318] [LOG]   Starting MagicMirror: v2.26.00
[01.01.2024 22:22.15.321] [LOG]   Running node 21.1.0 and npm 10.2.4 on "Linux debian 5.10.0-20-amd64 #1 SMP Debian 5.10.158-2 (2022-12-13) x86_64 GNU/Linux"
```

What do you think? Are there other information also be helpful?

Maybe also something to check if the user really did `npm install` for an update and not just `git pull`.

khassel, 2024-01-01:
> Maybe also something to check if the user really did npm install for an update and not just git pull.

at the moment I have no idea how to check this (but it would be helpful)

> Are there other information also be helpful?

these are my wishes:

```bash
magic_mirror on  develop [✘!?] is 📦 v2.27.0-develop via  v21.5.0 took 43s
⬢ [Docker] ❯ uname -a
Linux ada7cbb69aa5 5.15.133.1-microsoft-standard-WSL2 #1 SMP Thu Oct 5 21:02:42 UTC 2023 x86_64 GNU/Linux

magic_mirror on  develop [✘!?] is 📦 v2.27.0-develop via  v21.5.0
⬢ [Docker] ❯ node -v
v21.5.0

magic_mirror on  develop [✘!?] is 📦 v2.27.0-develop via  v21.5.0
⬢ [Docker] ❯ npm -v
10.2.4

magic_mirror on  develop [✘!?] is 📦 v2.27.0-develop via  v21.5.0
⬢ [Docker] ❯ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

magic_mirror on  develop [✘!?] is 📦 v2.27.0-develop via  v21.5.0
⬢ [Docker] ❯ id
uid=1000(node) gid=1000(node) groups=1000(node),44(video),997(gpio)

magic_mirror on  develop [✘!?] is 📦 v2.27.0-develop via  v21.5.0
⬢ [Docker] ❯ ls -la / | grep .dockerenv
-rwxr-xr-x   1 root root    0 Jan  1 23:08 .dockerenv
```

sdetweil, 2024-01-01:
you can check the version of electron or node-ical.in the node_modules folder. we know those changed to release

sdetweil, 2024-01-02:
we still have some users running hybrid mode, 64 bit kernel, 32 it runtime

file $(which bash)
/bin/bash: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib64/ld-linux-x86-64.so.2, BuildID[sha1]=2a9f157890930ced4c3ad0e74fc1b1b84aad71e6, for GNU/Linux 3.2.0, stripped

will output the type of runtime binary format

closed 2024-04-01 by rejas