MagicMirrorOrg/MagicMirror #3329: Calendar module: TypeError: fetch failed

mattcd216, 2024-01-02:
**Platform**: 
`pi@magicmirror:~ $ uname -a
Linux magicmirror 6.1.21-v7+ #1642 SMP Mon Apr  3 17:20:52 BST 2023 armv7l GNU/Linux
pi@magicmirror:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye`

**Node Version**: v20.10.0

**MagicMirror² Version**: v2.26.0

**Description**: My calendar module does not appear to update at all since the first time it loaded. No events created since the calendar has been added to the module are showing.

**Steps to Reproduce**: 
1. Add my calendar server & credentials to module config
2. pm2 restart MagicMirror
3. View calendar successfully on mirror
4. Add an event to my calendar which will show within the maximumEntries events
5. Wait period to allow for fetch interval

**Expected Results**:  I would expect to see the new event I created, especially as I can see events occurring later than my new event. 

**Actual Results**: There is no change to the events shown on the mirror. No new events created are showing. Tried with several new events that are expected to show. The calendar module is still showing events in 3 days time but will not show my newly created event that is scheduled for tomorrow. 
Errors noted:
`pi@magicmirror:~/.pm2/logs $ tail --lines=1000 MagicMirror-error.log | grep calendar
[02.01.2024 10:29.43.758] [ERROR] Calendar Error. Could not fetch calendar:  https://myurl TypeError: fetch failed
[02.01.2024 10:31.53.769] [ERROR] Calendar Error. Could not fetch calendar:  https://myurl TypeError: fetch failed
`

**Configuration**: 
```
{
module: "calendar",
header: "home cloud calendar",
position: "top_left",
config: {
	fetchInterval: 120000,
	timeFormat: 'absolute',
	maximumEntries: 5,
	calendars: [
		         {
			url: 'https://my.locally.hosted.baikal.server.url/dav.php/user/calendar?export',
			auth: 	{
				user: 'user',
				pass: 'pass',
				method: 'basic'
		         		}
			}
			],
		}
},
```

**Additional Notes**: pm2 restart or a full reboot do not resolve

sdetweil, 2024-01-02:
could you do 
```
tail --lines=1000 MagicMirror-error.log | grep -i stack 
```
to see if there is a stack trace in there

mattcd216, 2024-01-02:
> could you do
> 
> ```
> tail --lines=1000 MagicMirror-error.log | grep -i stack 
> ```
> 
> to see if there is a stack trace in there

Nothing showing, no

mattcd216, 2024-01-02:
[02.01.2024 14:59.23.139] [ERROR] Calendar Error. Could not fetch calendar:  https://my.cal.url TypeError: fetch failed
    at Object.fetch (node:internal/deps/undici/undici:11576:11)
    at process.processTicksAndRejections (node:internal/process/task_queues:95:5) {
  cause: ConnectTimeoutError: Connect Timeout Error
      at onConnectTimeout (node:internal/deps/undici/undici:8522:28)
      at node:internal/deps/undici/undici:8480:50
      at Immediate._onImmediate (node:internal/deps/undici/undici:8511:13)
      at process.processImmediate (node:internal/timers:476:21) {
    code: 'UND_ERR_CONNECT_TIMEOUT'
  }
}

Themaster312, 2024-01-02:
Same issue here

sdetweil, 2024-01-02:
@Themaster312   try this test code   replace line 58-84 with this 

```
		try {

		fetch(url, { headers: headers, agent: httpsAgent })
			.then(NodeHelper.checkFetchStatus)
			.then((response) => response.text())
			.then((responseData) => {
				let data = [];

				try {
					data = ical.parseICS(responseData);
					Log.debug(`parsed data=${JSON.stringify(data)}`);
					events = CalendarFetcherUtils.filterEvents(data, {
						excludedEvents,
						includePastEvents,
						maximumEntries,
						maximumNumberOfDays
					});
				} catch (error) {
					fetchFailedCallback(this, error);
					scheduleTimer();
					return;
				}
				this.broadcastEvents();
				scheduleTimer();
			})
			.catch((error) => {
				fetchFailedCallback(this, error);
				scheduleTimer();
			});
		}
		catch(error){
                    console.error( "fetch for calendar url" +url + "failed error="+error.code);
                    fetchFailedCallback(this, error);
                    scheduleTimer();
		}
```

khassel, 2024-01-02:
The `fetch` call didn't change between `v2.25.0` and `v2.26.0`.

Found [this mentioned as workaround](https://undici.nodejs.org/#/?id=network-address-family-autoselection) but not sure if it is really related.

sdetweil, 2024-01-02:
@Pollard-J is this the issue you were looking for?

Pollard-J, 2024-01-02:
@sdetweil  The issue i had with my calendar was all to do with the Node needing to be updated. which you helped me with. thats all fixed now.

I did just update my MM and it broke it (it wouldn't launch). I updated the pm2 and its fixed it.

sdetweil, 2024-01-02:
@Pollard-J   thanks for the update.

sdetweil, 2024-01-03:
@Pollard-J 

did u upgrade with my script? or manually (git pull && ... ) ??

sdetweil, 2024-01-03:
@khassel 

>but not sure if it is really related

but that seems to be on request() and userAgent() methods, not fetch()..

he did a good upgrade w my script. (from v23 to 26) also upgraded node from 16 to 20. 

he has a repeatable failure..

khassel, 2024-01-03:
so I have no idea to this anymore, only a few users are affected and we don't know why ...

One last thing: Is mm started with `npm run server` or `npm run start`?

If they are using electron (`npm run start`) they are running on the node version provided by electron which is with current release `18.17.1` so upgrading local nodejs is useless.

It would be interesting to see if this problem persists when installing the latest electron version.

sdetweil, 2024-01-03:
these are npm run start... using electron.. 

but he jumped a bunch of releases.. 23-26. 

I think the node_helpers are running under command line node.. as electron. loadUrl()  hasn't been started yet.  as you note electron has its own..

bugsounet, 2024-01-04:
Hi, I have 2 users with same problem
callback is not sended and break some others modules

sdetweil, 2024-01-04:
that's different. other side closed.

bugsounet, 2024-01-04:
yes but callback is not sended too

sdetweil, 2024-01-05:
what callback,?  you mean the .then() Promise resolve?

GaryCooper23, 2024-01-06:
`# Disable IPv6
net.ipv6.conf.all.disable_ipv6 = 1
net.ipv6.conf.default.disable_ipv6 = 1
net.ipv6.conf.lo.disable_ipv6 = 1
`

Solved it for now. Please implement a better solution.

sdetweil, 2024-01-06:
@GaryCooper23  thanks for the feedback. the problem is not in our code. but in the node js engine.  we will have to wait for a fix.

 but you don't need ip 6 on your local network, as you only have a few devices.

khassel, 2024-01-06:
Maybe the problems are caused by the node version (`18.17.1`) in the used electron version (`27.2.0`).

The were fixed for `autoSelectFamily` with node version `18.18.0`.

The latest electron release `28.1.1` contains the node version `18.18.2`, so contains the above fixes.

It would be helpful for us if someone with this problem could test if it still persists when using electron `28.1.1`.

sdetweil, 2024-01-06:
ok  will see

sdetweil, 2024-01-09:
one feedback that 28.1.1 solved problem

sdetweil, 2024-02-06:
@z-master42 install the old parser

```
cd ~/MagicMirror
npm install node-ical@0.16.1
```

restart MagicMirror

z-master42, 2024-02-06:
`node-ical` is on `0.16.1`. No change so far.

khassel, 2024-07-01:
is this still an issue with latest MagicMirror Release v2.28.0?

Background: In the new release we are using electron v31 which internally uses node v20 (instead of v18 before).

k0b4lt, 2024-07-15:
> is this still an issue with latest MagicMirror Release v2.28.0?
> 
> Background: In the new release we are using electron v31 which internally uses node v20 (instead of v18 before).

Seems to be. Just did a fresh install using the docker image provided by karsten13/magicmirror:fat which lists v2.28.0 as the version used. Exact same error for the newsfeed module (obviously referncing newsfeed:  https://rss.nytimes.com/services/xml/rss/nyt/HomePage.xml though)

```
[2024-07-15 07:27:48.330] [ERROR] Calendar Error. Could not fetch calendar:  https://ics.calendarlabs.com/76/mm3137/US_Holidays.ics TypeError: fetch failed
at node:internal/deps/undici/undici:12502:13
at processTicksAndRejections (node:internal/process/task_queues:95:5)
at runNextTicks (node:internal/process/task_queues:64:3)
at process.processImmediate (node:internal/timers:449:9) {
[cause]: ConnectTimeoutError: Connect Timeout Error
at onConnectTimeout (node:internal/deps/undici/undici:6635:28)
at node:internal/deps/undici/undici:6587:50
at Immediate._onImmediate (node:internal/deps/undici/undici:6619:13)
at process.processImmediate (node:internal/timers:478:21) {
code: 'UND_ERR_CONNECT_TIMEOUT'
}
}
```

z-master42, 2024-07-15:
Everything works for me. From my point of view, this issue can be closed.

k0b4lt, 2024-07-15:
> > N/m this. seems to be something with the docker container only. spun up a virtual machine and followed the install steps on https://docs.magicmirror.builders/getting-started/installation.html#manual-installation and have no issues like this anymore
> 
> can you share the used `docker-compose.yml` file?

the one included in the repo for server only. I was trying it to test in a browser on a local machine... or attempting to at least

https://gitlab.com/khassel/magicmirror/-/blob/master/run/serveronly.yml?ref_type=heads

khassel, 2024-07-15:
so you are using `npm run server` for starting mm (no electron involved, but interesting anyway).

I still can't reproduce this error, you can try with the following, maybe it helps.

```yaml
version: '3'

services:
  magicmirror:
    container_name: mm
    image: karsten13/magicmirror:latest
    network_mode: host
    privileged: true
    volumes:
      - ../mounts/config:/opt/magic_mirror/config
      - ../mounts/modules:/opt/magic_mirror/modules
      - ../mounts/css:/opt/magic_mirror/css
    restart: unless-stopped
    command: 
      - npm
      - run
      - server
```

sdetweil, 2024-07-15:
but...did you change the MagicMirror config.js to allow outside system access.
by default it allows ONLY  apps INSIDE the same machine (docker container is machine) to access MagicMirror web page

address and ipWhitelist  need to be changed
docker doesn't know about app level restrictions

k0b4lt, 2024-07-15:
tried again at home just now on a rpi4. setup using portainer using the fat image again and it worked as expected. no errors from undici for either module

khassel, 2024-08-01:
seems to be solved with latest mm/electron version.

closed 2024-08-01 by khassel

Cam9731, 2024-08-26:
I seem to be having the same issue on Node version v20.9.0 and MagicMirror version 2.28.0.

Anyone else still experiencing this/having issues with this?

`0|MagicMir | [2024-08-26 11:22:20.728] [ERROR] Calendar Error. Could not fetch calendar:  http://www.calendarlabs.com/ical-calendar/ics/76/US_Holidays.ics TypeError: fetch failed
0|MagicMir |     at node:internal/deps/undici/undici:13178:13
0|MagicMir |     at processTicksAndRejections (node:internal/process/task_queues:95:5)
0|MagicMir |  {
0|MagicMir |   [cause]: ConnectTimeoutError: Connect Timeout Error
0|MagicMir |       at onConnectTimeout (node:internal/deps/undici/undici:2331:28)
0|MagicMir |       at node:internal/deps/undici/undici:2283:50
0|MagicMir |       at Immediate._onImmediate (node:internal/deps/undici/undici:2315:13)
0|MagicMir |       at processImmediate (node:internal/timers:483:21)
0|MagicMir |    {
0|MagicMir |     code: 'UND_ERR_CONNECT_TIMEOUT'
0|MagicMir |   }
0|MagicMir | }
`

sdetweil, 2024-08-26:
@Cam9731 that looks like a networking problem, connection timeout

Cam9731, 2024-08-26:
@sdetweil Hahaha looks like my pihole was misbehaving, sorry for the false alarm and thanks for the quick response! :)

sdetweil, 2024-08-26:
glad you found it