MagicMirrorOrg/MagicMirror #3333: MagicMirror cannot start with NodeJS 21.5.0 on Alpine Linux

spupetic, 2024-01-03:
**Platform**: Alpine Linux, armv7l, 6.6.0
**Node Version**: v21.5.0

**MagicMirror² Version**: 2.26.0

**Description**: 
I am getting this error when I run `npm run start`.
```
 magicmirror@2.26.0 start
> DISPLAY="${DISPLAY:=:0}" ./node_modules/.bin/electron js/electron.js

node:events:497
      throw er; // Unhandled 'error' event
      ^

Error: spawn /home/user/MagicMirror/node_modules/electron/dist/electron ENOENT
    at ChildProcess._handle.onexit (node:internal/child_process:286:19)
    at onErrorNT (node:internal/child_process:484:16)
    at process.processTicksAndRejections (node:internal/process/task_queues:82:21)
Emitted 'error' event on ChildProcess instance at:
    at ChildProcess._handle.onexit (node:internal/child_process:292:12)
    at onErrorNT (node:internal/child_process:484:16)
    at process.processTicksAndRejections (node:internal/process/task_queues:82:21) {
  errno: -2,
  code: 'ENOENT',
  syscall: 'spawn /home/user/MagicMirror/node_modules/electron/dist/electron',
  path: '/home/user/MagicMirror/node_modules/electron/dist/electron',
  spawnargs: [ 'js/electron.js' ]
}
```
**Steps to Reproduce**: Install MagicMirror on Alpine, and try to start it.

**Configuration**:  I am using the default config.

sdetweil, 2024-01-03:
can u back down to node 20.8.0 please.. that is a highest we have tested.. this is an electron problem ..

sdetweil, 2024-01-03:
also, ouptut of 

uname -a
lsb_release -a

sdetweil, 2024-01-03:
did you use my script or do it manually?

khassel, 2024-01-03:
tell us more about your setup, you are running alpine on a pi or which hardware?

Electron needs a graphical device, how did you do this with alpine?

khassel, 2024-01-03:
I can reproduce the above error if I use my alpine docker image (which is only meant to use as server only setup) and install electron and then run `npm start`.

What you want to achieve is possible but electron has no release for running under alpine, see [this issue](https://github.com/electron/electron/issues/9662) for more info.

So if you get electron running under alpine mm will also run.

spupetic, 2024-01-04:
Thanks for your information!

I have installed it manually, according to the installation and usage page from the documentation.

I am using postmarketOS edge on a Galaxy Ace 4, so downgrading nodejs is not easy, I have to compile it because Alpine does not keep more versions in it's repo and nvm is not working well on this distro.

khassel, 2024-01-04:
I will close this because your main problem is to get electron running.

If you are able to start an electron app e.g. https://github.com/electron/electron-quick-start then mm should run too.

closed 2024-01-04 by khassel

sdetweil, 2024-01-04:
but, you could run this as a server and access from somewhere else..  and not need electron