MagicMirrorOrg/MagicMirror #3347: Weather module, openweathermap, can't get more than 5 days forecast

puenteNicholas21, 2024-01-12:
## I found a bug in MagicMirror

**Platform**: raspi 2 b

**Node Version**: 18.19.0

**MagicMirror² Version**: 2.26.0

**Description**: Weather module using openweathermap does not return more than 5 days forecast. changed both maxnumberofdays and max entries and no luck. changed maxnumberofdays to 3 as a test that the variable is getting passed and it works just seems to not get anything over 5.

**Steps to Reproduce**: configure weather module to use openweathermap, forecast, maxnumberofdays > 5

**Expected Results**:  more than 5 day forecast output

**Actual Results**: max of 5 days forecast

**Configuration**: 
{
      module: "weather",
      position: "top_right",
      header: "Weather Forecast",
      config: {
        weatherProvider: "openweathermap",
        type: "forecast",
        maxNumberOfDays: 7,
        fade: false,
        location: "Wylie",
        locationID: "4743275", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
        apiKey: "<APIKEY>",
      },
    },

**Additional Notes**: other modules/configs working as expected

rejas, 2024-01-12:
Unfortunately the free API from openweathermap only allows a 5 day forecast. So nothing we can do codewise here. 
Of course this would be good to add to the Documentation (if it is not in there). Maybe this is something you want to help us with?

puenteNicholas21, 2024-01-12:
> Unfortunately the free API from openweathermap only allows a 5 day forecast (see https://openweathermap.org/api). So nothing we can do codewise here. Of course this would be good to add to the Documentation (if it is not in there). Maybe this is something you want to help us with?

Ok i was afraid that was the answer but makes sense. If i can figure out how to update the docs i will try and do so.

@rejas Curious if you know, i was looking at the api page and it says you get 1000 free calls a day. with the default configs is it making anywhere close to that? meaning if i do subscribe for a paid account i would ideally never be charged at the default rate calls are being made right? Also, i saw mention of moving to 3.0 on openweather docs so not sure if anyone looking at that at all?

closed 2024-01-12 by puenteNicholas21

NichoBridge214, 2024-01-12:
@rejas https://github.com/MagicMirrorOrg/MagicMirror-Documentation/pull/228

never done pull request so hopefully everything is set up correctly

rejas, 2024-01-12:
thanks, had a comment, so please rework