MagicMirrorOrg/MagicMirror #3352: Does anyone know how to solve this problem

Leo77777777777, 2024-01-15:
[image]
Hello and thank you for opening an issue.

**⚠️ Please make sure that you have read the following lines before submitting your Issue:**

## I'm not sure if this is a bug

If you're not sure if it's a real bug or if it's just you, please open a topic on the forum: [https://forum.magicmirror.builders/category/15/bug-hunt](https://forum.magicmirror.builders/category/15/bug-hunt)

## I'm having troubles installing or configuring MagicMirror

Problems installing or configuring your MagicMirror? Check out: [https://forum.magicmirror.builders/category/10/troubleshooting](https://forum.magicmirror.builders/category/10/troubleshooting)

A common problem is that your config file could be invalid. Please run in your MagicMirror² directory: `npm run config:check` and see if it reports an error.

## I found a bug in the MagicMirror² installer

If you are facing an issue or found a bug while trying to install MagicMirror² via the installer please report it in the respective GitHub repository:
[https://github.com/sdetweil/MagicMirror_scripts](https://github.com/sdetweil/MagicMirror_scripts)

## I found a bug in the MagicMirror² Docker image

If you are facing an issue or found a bug while running MagicMirror² inside a Docker container please create an issue in the corresponding repository:
[https://gitlab.com/khassel/magicmirror](https://gitlab.com/khassel/magicmirror)

## I'm having troubles installing or configuring foreign modules

Please open an issue in the module repository or ask for help in the [forum](https://forum.magicmirror.builders/)

---

## I found a bug in MagicMirror

Please make sure to only submit reproducible issues. You can safely remove everything above the dividing line.
When submitting a new issue, please supply the following information:

**Platform**: Place your platform here... give us your web browser/Electron version _and_ your hardware (Raspberry Pi 2/3/4, Windows, Mac, Linux, System V UNIX).

**Node Version**: Make sure it's version 18 or later (recommended is 20).

**MagicMirror² Version**: Please let us know which version of MagicMirror² you are running. It can be found in the `package.json` file.

**Description**: Provide a detailed description about the issue and include specific details to help us understand the problem. Adding screenshots will help describing the problem.

**Steps to Reproduce**: List the step by step process to reproduce the issue.

**Expected Results**: Describe what you expected to see.

**Actual Results**: Describe what you actually saw.

**Configuration**: What does the used config.js file look like? Don't forget to remove any sensitive information!

**Additional Notes**: Provide any other relevant notes not previously mentioned. This is optional.

rejas, 2024-01-15:
Have you read the text below the image in the issue template above?

⚠️ Please make sure that you have read the following lines before submitting your Issue:

Leo77777777777, 2024-01-18:
I'm still at school, yes I've read the lines but I don't really understand what they mean, even though I've done a lot of research 

> Você leu o texto abaixo da imagem no modelo de edição acima?
> 
> ⚠️Certifique-se de ter lido as seguintes linhas antes de enviar seu problema:

Yes, I tried reading it and I didn't understand it very well, I'm still at school and that's why I came here to ask for help from someone who understands.

rejas, 2024-01-18:
Ok, please go to the forum (https://forum.magicmirror.builders) and ask your question there. This doesnt seem like a bug that needs to be reported here.

closed 2024-01-18 by rejas