MagicMirrorOrg/MagicMirror #3353: logLevel before loading config.js

khassel, 2024-01-15:
When debugging another issue i stumbled over the `[DEBUG]` line in the start logs when using default config:

```bash
> magicmirror@2.27.0-develop server
> node ./serveronly

[15.01.2024 20:25.31.730] [LOG]   Starting MagicMirror: v2.27.0-develop
[15.01.2024 20:25.31.749] [LOG]   Loading config ...
[15.01.2024 20:25.31.749] [DEBUG] config template file not exists, no envsubst
[15.01.2024 20:25.32.191] [LOG]   Loading module helpers ...
[15.01.2024 20:25.32.192] [LOG]   No helper found for module: alert.
[15.01.2024 20:25.32.196] [LOG]   Initializing new module helper ...
```

In `js/app.js` we are loading the config and using `Log.debug`, `Log.warn` and `Log.error`.
The `LogLevel` defined in the config is set **after** config is loaded, so the above Log statements are running without setting a logLevel before which means all logLevel are displayed including `DEBUG`.

We could do different things to solve this
- simple solution: We accept this behavior and replace `Log.debug` with `Log.info` for the few lines
- big solution: We do not log before config is loaded but we collect the messages in several lists (debugList, infoList, ...) and print the output after config is loaded (then we know what List to print and which not).

Or other ideas? @rejas @sdetweil @KristjanESPERANTO

KristjanESPERANTO, 2024-01-16:
Good point. At the moment, I think that the simple solution would be sufficient :slightly_smiling_face:

rejas, 2024-01-16:
I too would go with the simple solution

closed 2024-04-01 by rejas