MagicMirrorOrg/MagicMirror #3358: cleanup file headers

khassel, 2024-01-16:
See discussion [here](https://github.com/MagicMirrorOrg/MagicMirror/pull/3350#discussion_r1451981420).

Beside the author headers: Need we in every file the `MIT Licensed` stuff?

Any other cleanup wishes?

khassel, 2024-01-16:
For the author headers: `find ./ -type f -iname *.js -exec sed -i -r '/^\s\* By.*$/d' {} \;`

KristjanESPERANTO, 2024-01-16:
In my understanding of MIT, it is not necessary to mention the license in all files. But I think it's fine to do it.

As a basis for discussion, here is the current header from `app.js`:

```js
   /* MagicMirror²
   * The Core App (Server)
   *
   * By Michael Teeuw https://michaelteeuw.nl
   * MIT Licensed.
   */
 ```
 
And here three spontaneous proposals:
 
  **Proposal 1:**
  
 ```js
   /*
    * MagicMirror²
    * The Core App (Server)
    *
    * MIT Licensed.
    */
 ```
 
 **Proposal 2:**
 
  ```js
   /*
    * MagicMirror²
    * The Core App (Server)
    *
    * This source code is licensed under the MIT license found in the
    * LICENSE file in the root directory of this source tree.
    */
 ```

 **Proposal 3:**
 
  ```js
   /*
    * MagicMirror²
    * The Core App (Server)
    */
 ```
 
 Feel free to tear these proposals apart. The license part in proposal 2 I found in the sources of [react](https://github.com/facebook/react/blob/main/scripts/eslint/index.js).

KristjanESPERANTO, 2024-01-16:
There are even a few ESLint plugins that check the headers. Here is the rule of one of them that seems suitable to me:: https://github.com/nikku/eslint-plugin-license-header/blob/master/docs/rules/header.md

The example there looks very similar to proposal 2.

```js
  /**
   * Copyright (c) Foo Corp.
   *
   * This source code is licensed under the WTFPL license found in the
   * LICENSE file in the root of this projects source tree.
   */
```

rejas, 2024-01-18:
Proposal 4: Remove headers alltogether. They provide no real information. I would rather have proper jsdoc headers everywhere ;-)

khassel, 2024-01-18:
from my side we can remove all these headers ...

KristjanESPERANTO, 2024-01-18:
I'm also fine with removing :+1:

rejas, 2024-03-21:
so I guess this is done and can be closed @khassel ?

closed 2024-03-21 by khassel