MagicMirrorOrg/MagicMirror #3380: default calendar losing events between backend and front end

sdetweil, 2024-02-02:
see

https://forum.magicmirror.builders/topic/18325/calendar-not-updating/85?_=1706893005707

I have added  traces to just before the socketsend from node helper and traces to the front end, and the event shows before send, and does not show after socket receive

the event is NOT at the start or end of the data

how else can I debug this?

we just did a new npm install, on 2.26, with node-ical@0.16.1

sdetweil, 2024-02-03:
socketio uses json stringify before send, and parse on receive

my traces do the same..

sdetweil, 2024-10-11:
closing , can't reproduce

emlowe, 2025-01-28:
I appear to be having this issue:

I added various log outputs to node_helper.js and calendar.js

node_helper is sending the socket notification for CALENDAR_EVENTS
this is not received by the front-end code in calendar.js

I don’t know what happens to it, but I added logging to print out in calendar.js whenever the socketNotificationReceived handler is called and it doesn’t print.

(unless I refresh the page - at which point it does)

emlowe, 2025-01-28:
I have seen this behavior in both server mode and with normal electron mode. Most of my recent debugging is in server mode so I can bring up the webpage in a different browser on a second machine

Current test environment:

rpi3 with Debian GNU/Linux 12 (bookworm)
Running develop branch of MM with node v22.13.0
Front-end on Brave browser on a mac

emlowe, 2025-01-29:
I'll note that the events do seem to eventually appear , although sometimes this takes several hours.

emlowe, 2025-01-29:
Same behaviour with Safari and Firefox

sdetweil, 2025-02-05:
any idea how to debug this? is socketio buffered?

sdetweil, 2025-02-08:
maybe the buffer size isn't big enough

change server.js 
add the line marked below
```js

			const io = socketio(server, {
				cors: {
					origin: /.*$/,
					credentials: true
				},
				allowEIO3: true,                // add this trailing comma
				maxHttpBufferSize: 1e7   // add this line, set buffer to 10 meg max from 1 meg
			});
```

emlowe, 2025-02-10:
I tried this and it didn't seem to make any difference (I tried 1e8 since that is the WS default). I also added some error logging

```                
socket.on("error", (err) => {
	console.error(`Socket error: ${err.message}`);
});
```

but nothing gets logged so it doesn't seem there is an error being thrown.

sdetweil, 2025-02-10:
thanks. the default i saw was 1e6, 1meg

as i understand it there is no error reported its just lost

https://socket.io/docs/v4/troubleshooting-connection-issues/

other thing was to changevtyevtimeout

sdetweil, 2025-02-10:
hit button on phone

KristjanESPERANTO, 2025-09-30:
Release 2.33.0 is out 🙂

PR #3862 (2025-08-28): fix for #3380, socket.io timeout closure
socket.io times out and closes the client side socket without any callback
sendSocntNotification() from the server side data is lost as the socket is closed.  but the client doesn't know

increase the timeout 

fixes #3380
commit eb719429 (2025-08-28): fix for #3380, socket.io timeout closure (#3862)

socket.io times out and closes the client side socket without any
callback
sendSocntNotification() from the server side data is lost as the socket
is closed. but the client doesn't know

increase the timeout 

fixes #3380
closed 2025-09-30 by KristjanESPERANTO