MagicMirrorOrg/MagicMirror #3381: 2.24.0 - Calendar Error. Could not fetch calendar:  http://0.0.0.0:8080/modules/calendars/fussball.ics TypeError: fetch failed

screensaverproblem, 2024-02-07:
Is work before update from 2.19 to 2.26.0

Platform is "electron": "^27.2.0 on Raspberry Pi 4 Model B Rev 1.2                                                                                                         
Linux mm 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux Raspbian GNU/Linux 10 (buster)
node version 18.18.0
MagicMirror² Version: "2.26.0"
The problem is 100% reproducible.

**Description**: I made a major update from my MMM with the script:
bash -c  "$(curl -sL https://raw.githubusercontent.com/sdetweil/MagicMirror_scripts/master/upgrade-script.sh)" ap
ply

Then i change my local ics path from localhost entry on 0.0.0.0 (I have raed there: https://github.com/MagicMirrorOrg/MagicMirror/issues/3147)
old:    url: "http://localhost:8080/modules/calendars/fussball.ics"
new:   url: "http://0.0.0.0:8080/modules/calendars/fussball.ics"

I check this with the command 
pi@mm:~/temp $ wget http://0.0.0.0:8080/modules/calendars/fussball.ics
--2024-02-07 18:39:12--  http://0.0.0.0:8080/modules/calendars/fussball.ics
Verbindungsaufbau zu 0.0.0.0:8080 … verbunden.
HTTP-Anforderung gesendet, auf Antwort wird gewartet … 200 OK
Länge: 11502 (11K) [text/calendar]
Wird in »fussball.ics« gespeichert.
fussball.ics                      100%[============================================================>]  11,23K  --.-KB/s    in 0s      
2024-02-07 18:39:12 (42,4 MB/s) - »fussball.ics« gespeichert [11502/11502]

ls -alt
drwxr-xr-x  2 pi pi  4096 Feb  7 18:39 .
-rw-r--r--  1 pi pi 11502 Feb  7 18:30 fussball.ics

check the data content with 
less fussball.ics 

BEGIN:VCALENDAR
VERSION:2.0
CALSCALE:GREGORIAN
PRODID:-//SabreDAV//SabreDAV//EN
X-WR-CALNAME:Fussball RBL
X-APPLE-CALENDAR-COLOR:#31CC7C
REFRESH-INTERVAL;VALUE=DURATION:PT4H
X-PUBLISHED-TTL:PT4H
BEGIN:VTIMEZONE
TZID:Europe/Berlin
BEGIN:DAYLIGHT
TZOFFSETFROM:+0100
TZOFFSETTO:+0200
TZNAME:CEST
DTSTART:19700329T020000
RRULE:FREQ=YEARLY;BYMONTH=3;BYDAY=-1SU
END:DAYLIGHT
BEGIN:STANDARD
TZOFFSETFROM:+0200
TZOFFSETTO:+0100
TZNAME:CET
DTSTART:19701025T030000
RRULE:FREQ=YEARLY;BYMONTH=10;BYDAY=-1SU
END:STANDARD
END:VTIMEZONE
BEGIN:VEVENT
DTSTAMP:20230803T085143Z
UID:0c54334b-a2c9-493c-96d0-8de8420ba498
SUMMARY:* RB Leipzig - Borussia Dortmund | Bundesliga | 31. Spieltag
... cut ...

logs from /home/pi/.pm2/logs/mm-error.log
[07.02.2024 19:26.12.449] [ERROR] Calendar Error. Could not fetch calendar:  http://0.0.0.0:8080/modules/calendars/fussball.ics TypeError: fetch failed
    at Object.fetch (node:internal/deps/undici/undici:11576:11) {
  cause: Error: read ECONNRESET
      at TCP.onStreamRead (node:internal/stream_base_commons:217:20) {
    errno: -104,
    code: 'ECONNRESET',
    syscall: 'read'
  }
}

check also with the command: tcpdump -ni lo port 8080 | grep ics
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on lo, link-type EN10MB (Ethernet), capture size 262144 bytes
19:25:21.770943 IP 127.0.0.1.40672 > 127.0.0.1.8080: Flags [P.], seq 1:243, ack 1, win 512, options [nop,nop,TS val 841418175 ecr 841418173], length 242: HTTP: GET /modules/calendars/personal_hw.ics HTTP/1.1

logs from /home/pi/.pm2/logs/mm-out.log
[07.02.2024 19:26.12.336] [INFO]  Calendar-Fetcher: Broadcasting 19 events from http://0.0.0.0:8080/modules/calendars/personal_hw.ics.

**Steps to Reproduce**: The problem is steady after every reboot or pm2 stop mm & pm2 start mm
**Expected Results**: Display calendar entries from my four calendars
**Actual Results**: Display of calendar entries are complete empty
**Configuration**: 
{
                                                symbol: "calendar-plus-o",
                                                url: "http://0.0.0.0:8080/modules/calendars/fussball.ics"                                      },

**Additional Notes**: various files see appendix

sdetweil, 2024-02-07:
http://0.0.0.0:8080/modules/calendars/fussball.ics

you had to create a new folder, right? cause the folders we provide are
```text
modules/default/calendar
```
not
```text
modules/calendars 
```

khassel, 2024-02-07:
looks like a duplicate of #3329 concerning the undici errors

screensaverproblem, 2024-02-08:
Thanks for the info. 

I had searched for the error in the forum, but only found #3147. I was probably missing the right search term.

sdetweil, 2024-02-08:
we don't replicate the issues to the forum in general

screensaverproblem, 2024-02-09:
Thanks for the info. 

I had searched for the error in the forum, but only found #3147. I was probably missing the right search term.

> http://0.0.0.0:8080/modules/calendars/fussball.ics
> 
> you had to create a new folder, right? cause the folders we provide are
> 
> ```
> modules/default/calendar
> ```
> 
> not
> 
> ```
> modules/calendars 
> ```

Yes, I create a new folder modules/calendars. But this is not the problem. When I move all *.ics to  modules/default/calendar and change the config.js, the error is still present.

sdetweil, 2024-02-09:
@screensaverproblem it was just a housekeeping issue to make sure I understand..

I've never seen 0.0.0.0 in a URL before, as its intended for a listening value, not a connection value.  

i would only use  localhost (meaning this system) or the local system name or ip address of AN interface

note that 0.0.0.0 means ALL interfaces (assume wifi and ethernet both active) , but that doesn't make sense on a connection request.. outbound on both?  
and also wifi for both ipv4 and ipv6.. both makes no sense.. and ONE is likely to to be rejected at the server side.. 
(and I would expect the libraries would clean up any ambiguity this causes by rejecting it eventually)

screensaverproblem, 2024-02-09:
> it was just a housekeeping issue to make sure I understand..
>
> I've never seen 0.0.0.0 in a URL before, as its intended for a listening value, not a connection value.
>
> i would only use localhost [...]

I have also tried the local ipv4 address 192.168.x.x instead of 0.0.0.0. The error remained identical.

sdetweil, 2024-02-09:
odd still.. I use http://localhost:mm_port  for all my temp calendars.. on all my machines with 2.26 and don't see this

screensaverproblem, 2024-02-09:
> odd still.. I use http://localhost:mm_port for all my temp calendars.. on all my machines with 2.26 and don't see this

I had also used the http://localhost:mm_port syntax before the major upgrade. Sometimes 3 entries of the first calendar were still displayed. 
However, I was still in the learning phase (where do I find log files and so on ...) so I don't know whether the error code was identical. 

The error is unexpected as I have been using my Magic Mirroro Proket for years without any major problems. 

I searched the forum again for the **[UUID](undici:11576)** and found this issue: 
Calendar loading randomly fails if files stored locally #3365 
-> But the solution approach sounds complicated and I don't fully understand it either. I'm just not a software developer ;-)

sdetweil, 2024-02-09:
> But the solution approach sounds complicated and I don't fully understand it either

welcome to the fun.. we don't either.. lol

none of this is our locally written code..  so, we're not sure what the issues are.. lots of changed things all at once. (no choice) should have been MORE stable, but we got WAY less.. 

another test approach would be to start MM in server mode (not using electron) 
npm run server

and then open a browser over it. http://localhost:mm_port

and see how it behaves..  node 18 and 20 have the builtin fetch like in electron, same libs..

khassel, 2024-07-01:
is this still an issue with latest MagicMirror Release v2.28.0?

Background: In the new release we are using electron v31 which internally uses node v20 (instead of v18 before).

khassel, 2024-08-01:
seems to be solved with latest mm/electron version.

closed 2024-08-01 by khassel