MagicMirrorOrg/MagicMirror #3385: Bug: Compliments module has `anytime` array mandatory

Feedy88, 2024-02-23:
If the compliments module is configured to only specific dates (e.g. Birthdays, Anniversaries), the Mirror will freeze on dates that are not configured to show a compliment. The freeze happens based on the time set for `updateInterval`. This forces to also configure `anytime` to bypass the issue.

rejas, 2024-03-14:
sorry for not answering sooner. cant reproduce your error with a config like this:
```
{
  module: "compliments",
  position: "top_bar",
  config: {
    compliments: {
      "....-03-14": ["Happy Birthday ❤️", "Happy Birthday 🌻"]
    }		
  }
},
```

how does your config look?

sdetweil, 2024-03-14:
the code should fail, as this line of code is always executed
line 102 on release 2.26
```
                // Add compliments for anytime
                Array.prototype.push.apply(compliments, this.config.compliments.anytime);  //< --- anytime doesnt exist
```

Feedy88, 2024-03-14:
@rejas it works if it find an entry for the current day. Try changing to "....-03-15" and the issue should be reproducible.

rejas, 2024-03-14:
Nope, still cant reproduce any error with this.
a) please describe "will freeze" a little more. do you errors on the console?
b) what is your config exactly? (minus private data)
c) are you using the electron version or are you using a browser to display the compliment module?

Feedy88, 2024-03-22:
a) All time-relevant modules will freeze and no longer rotate (Clock seconds are no longer increasing, newsfeed module is stuck and does not rotate, Pir module does not count down and does not turn off the screen...)

b) I created a test-confing only using default modules and can reproduce the issue with that:
`/* Magic Mirror Config Sample
 *
 * By Michael Teeuw https://michaelteeuw.nl
 * MIT Licensed.
 *
 * For more information on how you can configure this file
 * See https://github.com/MichMich/MagicMirror#configuration
 *
 */

var config = {
	address: "localhost", 	// Address to listen on, can be:
				// - "localhost", "127.0.0.1", "::1" to listen on loopback interface
				// - another specific IPv4/6 to listen on a specific interface
				// - "0.0.0.0", "::" to listen on any interface
				// Default, when address config is left out or empty, is "localhost"
	electronOptions: {
		webPreferences: {
			webviewTag: true
		}
	},
	port: 8080,
	basePath: "/", 	// The URL path where MagicMirror is hosted. If you are using a Reverse proxy
			// you must set the sub path here. basePath must end with a /
	ipWhitelist: ["127.0.0.1", "::ffff:127.0.0.1", "::1","localhost", "192.168.178.0/24"], 	// Set [] to allow all IP addresses
								// or add a specific IPv4 of 192.168.1.5 :
								// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.1.5"],
								// or IPv4 range of 192.168.3.0 --> 192.168.3.15 use CIDR format :
								// ["127.0.0.1", "::ffff:127.0.0.1", "::1", "::ffff:192.168.3.0/28"],
	zoom: 1.0,
	useHttps: false, 		// Support HTTPS or not, default "false" will use HTTP
	httpsPrivateKey: "", 	// HTTPS private key path, only require when useHttps is true
	httpsCertificate: "", 	// HTTPS Certificate path, only require when useHttps is true

	language: "de",
	logLevel: ["INFO", "LOG", "WARN", "ERROR"],
	timeFormat: 24,
	units: "metric",
	// serverOnly:  true/false/"local" ,
	// local for armv6l processors, default
	//   starts serveronly and then starts chrome browser
	// false, default for all NON-armv6l devices
	// true, force serveronly mode, because you want to.. no UI on this device

	modules: [
		{
			module: "newsfeed",
			position: "bottom_bar",
			config: {
				feeds: [
					{
						title: "Tagesschau",
						url: "https://www.tagesschau.de/index~rss2.xml",
					},
					{
						title: "Heise",
						url: "https://www.heise.de/rss/heise-top-atom.xml",
					},
				],
				showSourceTitle: true,
				showPublishDate: true,
				ignoreOldItems: true,
				ignoreOlderThan: 86400000,
			}
		},
		{
			module: "alert",
		},
		{
			module: "updatenotification",
			position: "top_bar"
		},
		{
			module: "clock",
			position: "top_left",
			config: {
				clockBold: false,
				showWeek: true,
				showSunTimes: true,
				lat: 51,
				lon: 7,
			}
		},
		{
			//TO DO: check customEvents for color-coding
			module: "calendar",
			header: "Termine & Feiertage",
			position: "top_left",
			config: {
				fetchInterval: 300000, //5 Minutes
				displaySymbol: true,
				showLocation: false,
				fade: true,
				fadePoint: 0.4,
				dateFormat: "D. MMM HH:mm",
				fullDayEventDateFormat: "D. MMM",
				timeFormat: "absolute",
				getRelative: 12,
				urgency: 0,
				maximumEntries: 10,
				displayRepeatingCountTitle: true,
				hideOngoing: false,
				nextDaysRelative: true,
				coloredSymbol: true,
				coloredText: true,
				calendars: [
					{
						symbol: "cocktail",
						url: "webcal://i.cal.to/ical/61/nrw/feiertage/77c4d7d1.8cce2b06-ca7ba139.ics",
						color: "#a368db"
					},
					{
						symbol: 'birthday-cake',
						url: 'webcal://localhost:8080/modules/static/birthdays.ics',
						color: "#ffcf40"
					},
				]
			}
		},
		{
			module: "compliments",
			position: "lower_third",
			config: {
				updateInterval: 30000,
				compliments: {
					"....-01-01": [
						"Frohes Neues Jahr!"
					],
					"....-02-23": [
						"123435"
					],
					"....-03-26": [
						"Happy Birthday XXX!"
					],
					"....-09-16": [
						"1234"
					],
					"....-11-05": [
						"1234!"
					]
				}
			}
		},

	]
};

/*************** DO NOT EDIT THE LINE BELOW ***************/
if (typeof module !== "undefined") {module.exports = config;}
`

c) I am using the electron version

rejas, 2024-03-22:
Thx, found the culprit, it was the randomIndex method :-) Will do a PR / Fix later today!

PR #3402 (2024-03-23): Fix compliments module bringing mirror to a halt
... when no compliments are to be displayed. We shouldnt even try to randomize when the array has no elements...

Fixes #3385
commit 57549fa1 (2024-03-23): Fix compliments module bringing mirror to a halt (#3402)

... when no compliments are to be displayed. We shouldnt even try to
randomize when the array has no elements...

Fixes #3385
closed 2024-04-01 by rejas