MagicMirrorOrg/MagicMirror #3398: Move Module MMM-awesome-alexa to abandoned modules

tfischer4765, 2024-03-09:
Module MMM-awesome-alexa hasn't had any commits, issues answered or the documentation updated in over a year. The documentation currently available does not work any more. 

I therefore suggest moving the module to the "abandoned" section.

KristjanESPERANTO, 2024-03-09:
Done: https://github.com/MagicMirrorOrg/MagicMirror/wiki/3rd-Party-Modules/_compare/b1db6c08eb5397af23f0c3f1362cd70aebde8c56...c8bad463cf7ee244697adf5b095aa386b7d90ce5

tfischer4765, 2024-03-09:
Thanks for helping out! I know I could edit it, but I'm a guest here and I am hesitant to stick my fingies into other people's business, potentially ignoring some established procedure and annoying people I may very well depend on very soon.

closed 2024-03-09 by tfischer4765