MagicMirrorOrg/MagicMirror #3409: MagicMirror draws things multiple times

TheEpicTen, 2024-04-01:
I'm adding 2 images of the clock and Newsfeed module to show what I mean.

https://imgur.com/a/2B9Sj6R

Not sure how to fix this or what's causing it, but I have the MMM-OnSpotify module, and it temporarily fixes every time the currently playing song changes.

KristjanESPERANTO, 2024-04-01:
Does the problem also occur if you deactivate all modules except for the clock?

TheEpicTen, 2024-04-01:
I'll have a look in about an hour.

TheEpicTen, 2024-04-01:
Well my mirror no longer starts. I have a solution, but not enough time to do it today. I'll work on it tomorrow

TheEpicTen, 2024-05-22:
So the only time this issue occurs is when something is playing and shown on the onspotify module. I'll put this issue on that repo.

Fabrizz, 2024-05-22:
I can reproduce this as it only happens on the machine that I use to develop. Im sure that its related on how chromium readers things (In this case I thing its related to css filters and opacity interactions).

I cannot find issues similar to this as some times MM2 appears drawn over other tab in my machine and that REALLY makes no sense.

Im sure that there is a way of using flags to change how rendering works in electron and or other browsers. Changing the viewport size fixes this as its just a rendering artifact.

Fabrizz, 2024-05-22:
[image]
Happens instantly with my setup

sdetweil, 2024-05-23:
I don't understand viewport?    MM IS the viewport.. full screen intended..

IF your module wants a smaller 'viewpport' then it has to use html idoms to create another , like iframe
https://developer.mozilla.org/en-US/docs/Web/CSS/Viewport_concepts 

else it uses size to control the drawing area height, width .... 

IF the module resurfaced from off screen/hidden, its resume() method would be called..

Fabrizz, 2024-05-23:
> I don't understand viewport?    MM IS the viewport.. full screen intended..
> 
> IF your module wants a smaller 'viewport' then it has to use html idoms to create another , like iframe
> https://developer.mozilla.org/en-US/docs/Web/CSS/Viewport_concepts 
> and use css to customize its size
> 
> else it uses size to control the drawing area height, width .... 
> 
> IF the module resurfaced from off screen/hidden, its resume() method would be called.. 

Nono, I'm just saying that a viewport change in the page (or a color change on the elements that have a filter) triggers a repainting, thus the bug disapears.

I made the corresponding bug report on the chromium bug tracker.

The best way (hacky way) to make this disapear would be to change the colors a bit over time or not use that background effect.

I don't know if maybe instead of using css for opacity using the alpha channel in the color would be different, as that is not applied after the render

sdetweil, 2024-05-23:
ok, we don't know about css when we do updateDom().. we check that the redendered content is the same or not. 
so if style changed under the covers (animated background) that would be a browser shadow dom change I think. and outside our knowledge

closed 2024-10-13 by github-actions[bot]