MagicMirrorOrg/MagicMirror #3419: [weather] unit of config is not respected on broadcasted notifications

eouia, 2024-04-02:
**Platform**: All (Mainly tested on MacOS, RPI 4)

**Node Version**: 20.8.0

**MagicMirror² Version**: 2.27

**Description**: When the default weather module broadcasts its notification as `WEATHER_UPDATED`, the `imperial` units of current / forecasting temperatures (and speed of wind) is not respected. it is delivered as `metric` value.

**Steps to Reproduce**:
```js
/* config/config.js*/
units: "imperial",

modules: [
             {
			module: "weather",
			position: "top_right",
			config: {
				weatherProvider: "openweathermap",
				type: "current",
				location: "New York",
				locationID: "5128581", //ID from http://bulk.openweathermap.org/sample/city.list.json.gz; unzip the gz file and find your city
				apiKey: "..."
			}
		},
  ...
```
**Expected Results**: 
[image]

**Actual Results**:

[image]

**Additional Notes**: I knew about this issue since very early MM, but this issue has yet to be reported.

skpanagiotis, 2024-07-26:
Hello, I would like to contribute with a solution.

sdetweil, 2024-07-26:
@skpanagiotis   ok, code it up and submit a pull request to be considered.

skpanagiotis, 2024-08-04:
I have add this PR #3519 for the issue.

closed 2024-09-30 by khassel