MagicMirrorOrg/MagicMirror #3420: online doc initial page on android phone hamburger menu doesnt respond

sdetweil, 2024-04-02:
second page works ok

rejas, 2024-04-20:
works for me when opening the docs on chrome. what browser are you using?

sdetweil, 2024-04-20:
so I have narrowed this down to my new phone.. it is a samsung fold.  in single pane view it is ok
but on openm pane the left menu has no data

sdetweil, 2024-04-20:
[image]

sdetweil, 2024-04-20:
[image]

rejas, 2024-04-20:
oh, you mean the forum, not the docs...

sdetweil, 2024-04-20:
it also happened on the docs, but doesn't now..

rejas, 2024-04-21:
good to hear that it works now

closed 2024-04-21 by rejas