MagicMirrorOrg/MagicMirror #3421: short dayname with time there

skyzuma, 2024-04-04:
i am not a native en speaker ... this is an modification ask, not a bug

my next calendar event is Saturday at 15.30 (german: Samstag um 15.30 Uhr) ... both are very long, german is much longer but to long for the available cell ... its possible to short this dayname?

there is an thread in the forum (https://forum.magicmirror.builders/topic/18610/set-the-calendar-day-to-a-short-version?page=1) with the suggestion to open an issue ...

here https://momentjscom.readthedocs.io/en/latest/moment/04-displaying/01-format/ is the needed format i think i need: Day of Week > "ddd" and i think the "problem" is in calendar.js:455 > nextWeek "dddd" but idk if this the problerm ... also nextDay with "TOMORROW" is to long if an time there

i need "Sat at 15:30" instead of Saturday at 15.30 ... i need a short dayname of the week but **only** if a time there ... same with tomorrow and other long names with an time (in german tomorrow (morgen) is a way shorter)

skyzuma, 2024-04-20:
ivex changed line 455: from "nextWeek: "dddd"" (full dayname) ro "nextWeek: "ddd"" (short dayname) an restart mm but no changes :( iam not a programmer ... any1 can help?

sdetweil, 2024-04-20:
its only for full day events.. no time..  AND also timeFormat:"relative"

I changed mine to ddd and created an event for Monday and it stopped at that code and
produced 'Mon'

skyzuma, 2024-04-21:
events without time works good with "ddd" ... and how i can change it for events **with** time? same for tomorrow (with and without time)

an option in config.js is a good idea ... but i think this is not important enough for a change ...

sdetweil, 2024-04-21:
the code says the config.js dateFormat property will be used for this..

skyzuma, 2024-07-01:
its not, sry ... or i am blind ... show me plz :)

closed 2024-10-13 by github-actions[bot]