MagicMirrorOrg/MagicMirror #3422: [calendar] Wrong values on broadcasted fullday events with RRULE

eouia, 2024-04-09:
**Platform**: All

**Node Version**: v20.8.0

**MagicMirror² Version**: 2.27

**Description**: A repeated(BY RRULE) full-day event is shown as normal on screen, but it has the wrong value internally, so it delivers the wrong value on broadcasting as a notification. 

**Steps to Reproduce**:
1. Sample ics URL : https://calendar.google.com/calendar/ical/41bfece74ae469910f6414f6f0e6d7348bd1cd5eb442d3f415ade4388ea453f9%40group.calendar.google.com/public/basic.ics
2. It has two events, one for single fullday event, and the other for repeated fullday event. 
[image]

3. Those are displayed as a normal on the MM screen because Fullday event doesn't show the startTime. 
[image]

4. But inside of the logic, Repeated events have wrong parsed startTime.(Or at least it has wrong TZ information)
To check this;
[image]
>> modules/default/calendar/calendar.js line 417

This shows;
[image]
Repeated events don't start at 00:00. It starts at 02:00 I think it is due to Europe/Berlin Timezone (GMT+2)

Usually, it doesn't matter because full-day events do not show the exact time. But when the events are broadcasted, they deliver the wrong time value.

sdetweil, 2024-10-11:
can you check the results on my updated cal module when u get a chance

see this.. waiting on user test for submitting PR here and node-ical
all testcases in both run successfully..

here is a test version of the fixes for all kinds of date problems.

best to make a new folder and git clone there

git  clone https://github.com/sdetweil/MagicMirror
cd ~/MagicMirror
git checkout fixcaldates
npm run install-mm
copy your config.js and custom.css from the prior folder
and the non-default modules you have installed..

this ONLY changes the default calendar
but DOES ship an updated node-ical library too

if you need to fall back, just rename the folders around again so that
your original is called MagicMirror

all the testcases for node-ical and MagicMirror execute successfully.

the 'BIG' change here is to get the local NON-TZ dates for the
rrule.between()

all the checking and conversion code is commented out or not used
the node-ical fixes are for excluded dates (exdate) values being adjusted for DST/STD time.. waiting to submit that PR
more node-ical changes for TZ based dates

one fix in calendar.js for checking if a past date was too far back,
but it never checked to see IF the event date was in the past.. (before today) so it chopped off too many

and one change in calendarfetcher.js to put out a better diagnostic message of the parsed data.. (exdate was excluded cause JSON stringify couldn't convert the complex structure)

sdetweil, 2024-10-17:
here is an updated test version of the fixes for all kinds of calendar date problems.

NOTE: the changed branch name
NOTE: this used the node-cal@0.19.0 library UNCHANGED

best to make a new folder and git clone there

git clone https://github.com/sdetweil/MagicMirror
cd MagicMirror
git checkout fixcaldates2 // <------ note this is a changed branch name
npm run install-mm
copy your config.js and custom.css from the prior folder
and the non-default modules you have installed…

this ONLY changes the default calendar
but DOES ship an updated node-ical library too

if you need to fall back, just rename the folders around again so that
your original is called MagicMirror

all the testcases for node-ical and MagicMirror execute successfully.

the ‘BIG’ change here is to get the local NON-TZ dates for the
rrule.between()

all the checking and conversion code is commented out or not used
the node-ical fixes are for excluded dates (exdate) values being adjusted for DST/STD time… waiting to submit that PR

one fix in calendar.js for checking if a past date was too far back,
but it never checked to see IF the event date was in the past… (before today) so it chopped off too many

and one change in calendarfetcher.js to put out a better diagnostic message of the parsed data… (exdate was excluded cause JSON stringify couldn’t convert the complex structure)

I added the tests you all have documented

please re-pull and checkout the new branch (I deleted the old branch)
and npm run install-mm again

sdetweil, 2024-10-21:
and you just have to examine the fulldayeevent:true flag and ignore the time.. ALL UTC dates have time (00:00:00)
and the time changes when viewing an event from a diff timezone.. as full date (date-type'date') is local time only

sdetweil, 2024-10-21:
i updated the non-recurring content to match the recurring as well..

closed 2025-01-01 by sdetweil